jgromes/RadioLib #706: SX1262 Frequency Error

lvplummer, 2023-03-20:
I am using the great RadioLib with an ESP32C3 and SX1262.  All works great except the reported frequency error.  I get one good report and then wildly inaccurate results.  Here is an example:  -2241769.75.  I am not sure if this is a bug or pilot error.

Larry

jgromes, 2023-03-21:
Can you post some details about your setup? The code, details about your platfrom etc. - pretty much everything requested by the issue template.

Frequency error is experimental on SX126x, but seemed to be working for my use case (the SX126x_Receive/Receive_Interrupt examples).

lvplummer, 2023-03-22:
The hardware is two Heltec Wireless Stick V3 boards with ESP32S3 (not C3 as in earlier post) and SX1262.  

https://www.amazon.com/dp/B07S5XLP6Q?psc=1&ref=ppx_yo2ov_dt_b_product_details

I am running your transmit and receive with interrupt examples on the two boards.  The only changes were pins and LoRA config as follows:

SX1262 radio = new Module(8, 14, 12, 13);
  SPI.begin (9, 11, 10);
int  state = radio.begin(908.0, 500.0, 10, 5, 0x12, 10, 8);

I am using 0x12 for a sync word to be compatible with RadioHead which doesn't use a sync word but still transmits the default FSK 0x12 sync word.

Everything so far works as expected except for frequency error.  I like to use frequency error as a simple feed forward AFC.

I run the exact same setup except for pin differences on a homebrew ESP32 and SX1276 board using the appropriate interrupt version examples which produce correct frequency error reports.

It took me a while to get SX1276 and SX1262 interop because I didn't realize that your CRC on/off defaults were different.  Once both were turned on, all is good.

Larry

jgromes, 2023-03-22:
Thanks for providing the configuration, with that I was able to replicate the issue. Turns out that there was a problem in the calulcation that I failed to see previously, ebcause I was testing with quite precise TCXXO modules. Using SX1278 with a bad crystal forced the issue. It should be fixed in the latest commit, thanks for reporting!

PS:

> It took me a while to get SX1276 and SX1262 interop because I didn't realize that your CRC on/off defaults were different. Once both were turned on, all is good.

As far as I can tell, defaults should be the same for both SX127x and SX126x. I tested that with `SX1278::begin()` and `SX1268::begin()`, the two different modules can communicate. Is that not the case for you?

lvplummer, 2023-03-22:
Thanks for the update.

I am seeing the SX1276 default for CRC is off and for SX1262 default is
on.  When I force both to on, interop is good.  I don't have an SX1278 to
test.

When should I start looking for the next commit?

Larry

jgromes, 2023-03-22:
> I am seeing the SX1276 default for CRC is off and for SX1262 default is
on

Where exactly are you seeing this? Looking through the code, it actually seems that neither one explicitly sets CRC.

lvplummer, 2023-03-22:
Seeing is the wrong word.  Using the defaults doesn't work.  When I set the CRC to on for SX1276, it works.  When CRC is set to on for the SX1262, it still works.

jgromes, 2023-03-23:
@lvplummer FYI I explicitly enabled CRCs in the latest commits, so the defaults should match now.

jgromes, 2023-03-23:
Yes, I pushed it yesterday. If you go through the conversation (in GitHub issues) you will see the referenced commits fixing it.