jgromes/RadioLib #710: Unknown error -2 with eByte E22M900M2S

Tintin4000, 2023-03-23:
I am using the library (5.7.0) for the first time and face an issue with the ebyte module E22M900M22S connected ton ESP32 module (Adafruit Huzzah).

First I had to modify the example to include the GPIO pin connected to the reset pin of the module as such:

SX1262 lora = new Module(RFM1262_NSS, RFM1262_DIO1, RFM1262_RST, RFM1262_BUSY, *loraSPI);

I am using the default GPIO pin for the SPI connectivity.

I am getting the following message after activating the RADIOLIB_DEBUG:
SX126x not found! (1 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 00 00 00 ff 94 57 0d 80 f0 21 fb 3f 64 00 00 00 | .....W...!.?d...
Expected string: SX1261

After 10 attempts, the unknow error -2 is raised.

Looking into the code, the library is trying to access a registry at the address 0x0320, and looking into the reference document for the Semtech SX1262, there is no mention of that particular register address.  Obviously, as this is part of the module initialization, I can't go very much further.

Any help would be appreciated.

jgromes, 2023-03-23:
All SX126x devices contain a device string register, it's just not documented. It seems like the device is not returning menaingful data, which would suggest either a fault in the wiring, or some other hardware issue.

Tintin4000, 2023-03-24:
I have double check the hardware connection and it looks correct, but maybe the module is damaged?  I will try other libraries to see if I can check the hardware and eliminate the library or any other aspect.  Feel free to close the issue for now.  Thank you for your swift reply.

Tintin4000, 2023-03-25:
I have connected my ESP32 board over SPI to a BME680, simply to verify that a communication could be established over SPI and no problem.  So the issue is not on the ESP32, nor on the SPI (using the same pin as for the connection to the ebyte E22M) and same wire.  I have tried with 3 different E22-900M22S with the same result.  The string from returned for the address 0x0320 is identical on the 3 boards.  I will try again using a new ebyte module and lower the soldering temperature.

jgromes, 2023-03-25:
> As I am using the default pin on the ESP32 so I am expecting a normal behavior.

Different ESP32 boards may have different default SPI pins. Again, please post the entire code.

> I have connected my ESP32 board over SPI to a BME680, simply to verify that a communication could be established over SPI and no problem.

How did you verify this? Could you confirm the values read over the SPI match the expected?

>  I will try again using a new ebyte module and lower the soldering temperature?

Damage due to high soldering temperature is very unlikely, especially if the returned string is the same.

Tintin4000, 2023-03-26:
1. The ESP32 dev board that I am using is an Adafruit Huzzah ESP32 and do use the standard GPIO (CS=5, CLK=18, MISO=19,MOSI=23).
2. The expected output after connecting the BME680 is as per the example and the temp, humidity are normal.
3. I know, I can't get my head around it.

Code used...

`#include <Arduino.h>
#include <RadioLib.h>

// Change this to the GPIO connected to the RFM1262 NSS pin
#define RFM1262_NSS 5 // GPIO 33
// Change this to the GPIO connected to the RFM1262 DIO1 pin
#define RFM1262_DIO1 25 // GPIO 14
// Change this to the GPIO connected to the RFM1262 BUSY pin
#define RFM1262_BUSY 36 // GPIO 27
// Change this to the GPIO connected to the RFM1262 RST pin
#define RFM1262_RST 21 // GPIO 32

// ESP32 SPI class
SPIClass *loraSPI = new SPIClass(VSPI);
// SX1262 class
//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(RFM1262_NSS, RFM1262_BUSY, RFM1262_RST, RFM1262_DIO1, *loraSPI);

// The Ping message will be sent by the node
char pingMsg[] = "PING";
// The Pong message will be sent by the gateway RFM1262_DIO1
char pongMsg[] = "PONG";

long timeSinceLastPacket = 0;

void setup()
{
  // Start serial communication
  Serial.begin(115200);
  Serial.println("SX1262 initializing...");

  // set RF input to high
  pinMode(2, OUTPUT);
  pinMode(17, OUTPUT);
  digitalWrite(2, HIGH);
  digitalWrite(17, HIGH);

  // // Reset RFM1262
  // pinMode(RFM1262_RST, OUTPUT);
  // digitalWrite(RFM1262_RST, LOW);
  // delay(100);
  // digitalWrite(RFM1262_RST, HIGH);
  // delay(100);

  // Initialize LoRa
  int state = lora.begin(868.0, 125.0, 8, 5);
  if (state != RADIOLIB_ERR_NONE)
  {
    Serial.println("\nLoRa begin failed " + String(state));
  }
  lora.setOutputPower(10);
  lora.setTCXO(2.4);
  lora.setDio2AsRfSwitch(true);
}

void loop()
{
  Serial.println("Sending a PING");
  lora.transmit((uint8_t *)pingMsg, sizeof(pingMsg));

  // We wait only 2 seconds for a reply from the gateway
  time_t waitForGateway = millis();

  // Flag if a response was received
  bool gotReply = false;

  // Loop until we get a package or run into a timeout
  while (!gotReply)
  {
    // Check if data over LoRa is available
    int state = lora.receive((uint8_t *)pongMsg, sizeof(pongMsg));
    switch (state)
    {
    case RADIOLIB_ERR_NONE:
      Serial.println("---");
      digitalWrite(LED_BUILTIN, LOW); // Turn on status LED

      Serial.println("Got data package: " + String(pongMsg) + " RSSI: " + String(lora.getRSSI()));

      gotReply = true;

      break;
    case RADIOLIB_ERR_RX_TIMEOUT:
      // Serial.println("Receive timeout");
      break;
    case RADIOLIB_ERR_CRC_MISMATCH:
      Serial.println("CRC mismatch");
      break;
    default:
      Serial.println("Unknown error " + String(state));
      break;
    }

    if ((millis() - waitForGateway) > 2000)
    {
      // Timeout ==> Finish the waiting loop
      Serial.println("Timeout after " + String(millis() - waitForGateway));
      break;
    }
  }
  if (!gotReply)
  {
    // Didn't get response for 2 seconds
    digitalWrite(LED_BUILTIN, HIGH); // Turn off status LED
    Serial.println("Receive failed, is the PONG running?");
    // We will send the next PING immediately!
  }
  else
  {
    // We received a PONG, send the next PING after 500ms
    delay(500);
  }
}`

jgromes, 2023-03-26:
You don't seem to be initializing the SPI. Since you used non-default SPI, RadioLib will not be able to perform initialization for you, hence you have to call `loraSPI->begin();` before calling `lora.begin()`. You might have to supply the SPI pins to the SPI begin method, I'm not sure how that is handled on the Huzzah board.

Tintin4000, 2023-03-28:
It works!!!   You have been very reactive to the post, very much appreciated.  Thank you so much for your help.

jgromes, 2023-03-28:
Glad it's working now!

closed 2023-03-28 by jgromes