jgromes/RadioLib #718: Platformio STM32WL fails due to missing SubGhz

jgromes, 2023-04-02:
Originated in https://github.com/jgromes/RadioLib/issues/696#issuecomment-1491085110

I can confirm that building on Platformio for STM32WL(55JC1) will fail as it looks like the SubGhz library is missing. 

```
   12 | #include <SubGhz.h>
      |          ^~~~~~~~~~
compilation terminated.
.pio\libdeps\stm32wl\RadioLib\src\modules\SX126x\STM32WLx_Module.cpp:12:10: fatal error: SubGhz.h: No such file or directory
```

Checking the output in verbose mode, the include path for SubGhz is definitely missing, only SPI is included from the STM32duino package.

```
arm-none-eabi-g++
-o .pio\build\stm32wl\lib30d\RadioLib\modules\RFM9x\RFM96.cpp.o
-c 
-std=gnu++14 
-fno-threadsafe-statics 
-fno-rtti 
-fno-exceptions 
-fno-use-cxa-atexit 
-mcpu=cortex-m4 
-mthumb 
-Os 
-ffunction-sections 
-fdata-sections 
-nostdlib 
--param max-inline-insns-single=500 
-DPLATFORMIO=60106 
-DSTM32WLxx 
-DSTM32WLE5xx 
-DSTM32WLxx 
-DARDUINO=10808 
-DARDUINO_ARCH_STM32 
-DARDUINO_NUCLEO_WL55JC1 
-DBOARD_NAME=\"NUCLEO_WL55JC1\" 
-DHAL_UART_MODULE_ENABLED 
-DUSE_FULL_LL_DRIVER 
-DVARIANT_H=\"variant_NUCLEO_WL55JC1.h\" 
-DNDEBUG 
-DUSE_CM4_STARTUP_FILE 
-I.pio\libdeps\stm32wl\RadioLib\src 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\libraries\SPI\src 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\avr 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\OpenAMP 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32WLxx_HAL_Driver\Inc 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32WLxx_HAL_Driver\Src 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\STM32WLxx 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\open-amp\lib\include 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\libmetal\lib\include 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\virtual_driver 
-IC:\Users\user\.platformio\packages\framework-cmsis\CMSIS\Core\Include 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32WLxx\Include 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32WLxx\Source\Templates\gcc 
-IC:\Users\user\.platformio\packages\framework-cmsis\CMSIS\DSP\Include 
-IC:\Users\user\.platformio\packages\framework-cmsis\CMSIS\DSP\PrivateInclude 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\cores\arduino 
-IC:\Users\user\.platformio\packages\framework-arduinoststm32\variants\STM32WLxx\WL54JCI_WL55JCI_WLE4J(8-B-C)I_WLE5J(8-B-C)I 
.pio\libdeps\stm32wl\RadioLib\src\modules\RFM9x\RFM96.cpp
```

One thing I found odd was that the STM32 framework for Platformio was missing the SubGhz library in CMakeLists, and the library itself was missing a CMakeLists.txt file. Unfortunately adding both did not fix the issue.

cc @matthijskooijman @derekmpeterson

derekmpeterson, 2023-04-02:
Thanks for looking into it!  I began experiencing issues compiling (with no changes to my project) around the same time as the changes mentioned in this thread (https://community.platformio.org/t/library-ldf-mode-ignored-for-lib-deps/32639), but am not familiar enough with Platformio to understand what the resolution should be.  

It seemed odd that, as you mentioned, SPI is included, and appears to be coming in correctly from the platform package, as SubGhz should be.

I found that if I reference SubGhz.h directly in my project code, platformio recognizes it as a dependency of my project, but it's still unavailable while compiling RadioLib code.

I was able to unblock myself for now by explicitly forcing the header to be available at compile time, and the library itself is available at link time because my project depends on it directly
```
build_flags =
    -I"/Users/derek/.platformio/packages/framework-arduinoststm32/libraries/SubGhz/src"
```

matthijskooijman, 2023-04-03:
> STM32 framework for Platformio was missing the SubGhz library in CMakeLists, and the library itself was missing a CMakeLists.txt file

This would make sense, since the library was recently added (by me), targeting the Arduino environment which does not need cmake files (I did not actually realize that pio needs these and cannot compile arduino-compatible libraries as-is). But it seems at least the CMakeLists.txt file inside the library was [added in upstream stm32duino](https://github.com/stm32duino/Arduino_Core_STM32/commit/99cc110091d6686d80a47543521a0a7f74419d3c) (and released in 2.5.0, are you using that version?).

> I found that if I reference SubGhz.h directly in my project code, platformio recognizes it as a dependency of my project, but it's still unavailable while compiling RadioLib code.

That suggests that PIO does some automatic library detection similar to Arduino, but might approach it in a different way (Arduino builds a global pool of dependencies and puts them in the include path always, instead of doing this separately for the sketch and libraries, I think to keep things simple and predictable...).

No ideas on how to further approach this, though.

jgromes, 2023-04-10:
@matthijskooijman I installed the latest platformio version of https://registry.platformio.org/platforms/platformio/ststm32, listed as 15.6.0, not sure how that translates to versions of stm32duino.

Had some more time to dig into this, it turns out that it can be fixed by the following:

1. Moving the include of `<SubGHz.h>` from the STM32WL files into the global Module.h file.
2. Use the presence of of macro `STM32WLxx` to decide whether to include SubGhz or not. I also tried using `SUBGHZSPI_BASE` for this check, but it seems to be always undefined, at least in Module.h. Not sure why, as it looks like its working for generating Doxygen.

@matthijskooijman is there a particular reason why `SUBGHZSPI_BASE` was used instead of `STM32WLxx`?

jgromes, 2023-04-10:
@derekmpeterson I just pushed a commit fixing this, though you will probably have to wait until I publish a new version of the library on Platformio.

matthijskooijman, 2023-04-10:
> @matthijskooijman is there a particular reason why SUBGHZSPI_BASE was used instead of STM32WLxx?

I think I did that because it is more specific - this code can run on chips that have a SUBGHZ radio included, of which this define is an indication. In particular, this means that 1) this code could potentially work on (not currently existing) chips with a subghz radio that are not part of the stm32wl family, and 2) this code would be omitted for (not currently existing) stm32wl-family chips *without* a subghz radio. It's not a very big deal, though, checking for `STM32WLxx` would probably work just as well in practice.

jgromes, 2023-04-10:
@matthijskooijman thanks for the info - in that case I'll keep checking for `STM32WLxx` until something breaks. When that happens, feel free to link to this comment with "I told you so" ;)