jgromes/RadioLib #719: sx1278 invertIQ?

K4KDR, 2023-04-05:
Hello!

My environment is the Arduino IDE w/ Radiolib v5.7.0 installed.

I have a mystery LoRa signal recorded as an **I/Q file** for analysis.  When viewed in **Inspectrum**, the LoRa signal is obviously inverted:

[image]

... and after determining the LoRa parameters, was able to confirm that this is a valid (but inverted) LoRa signal by adding blocks in my GNU Radio 'Adalm Pluto Transmit' flowgraph to **swap** the I & Q before transmission over-the-air:

[image]

... with a **successful decode** using an Arduino Uno w/ Dragino sx1278 LoRa board.

In reviewing the radiolib code (please keep in mind my elementary understanding of these things), I found an '**invertIQ**' boolean flag in:

`RadioLib/src/modules/SX127x/SX127x.h`

... and "thought" that I could implement this by including the following line in my code:

`#define invertIQ              true`

... however, it has no effect.  Set to either true or false, if I transmit the test signal over-the-air in its normal state, I get NO decode.   If I transmit the test signal with I/Q reversed, I get a good decode regardless of whether I have set 'inverIQ' to either true or false.

Am I not implementing this feature correctly?   Or is there something else on this subject that I am simply not aware of?

Also, is there a command to **query** whether your device is configured for normal or inverted IQ?

Many thanks for any guidance!

-Scott,  K4KDR

K4KDR, 2023-04-05:
Oh my goodness!   Using the setting of the CRC parameter as a guide, I found my omission.  

In the appropriate sections of the sketch, in addition to setting TRUE or FALSE, I realized that a **'radio.invertIQ'** command had to be utilized as follows:

[image]

... hopefully someone will benefit by finding this in the ISSUE history of your repository here.  But, unless there is anything to add, no assistance is needed.   Please close the issue at your own descretion.

Many thanks as always & best regards,

-Scott

jgromes, 2023-04-05:
Glad you were able to resolve it! Out of curiosity, where is this mystery signal coming from?

closed 2023-04-05 by jgromes

K4KDR, 2023-04-05:
That's the problem - it appears to be coming from an as-yet-unknown satellite on an S-Band downlink just below 2200 MHz.

However, the 125 kHz bandwidth is not one of the standard parameters for 2.4 GHz LoRa, so I had to get creative and use my 433 MHz Dragino BEHIND a down-converter.   Now that it's decoding, perhaps we can learn something about the source.  Others have seen slow doppler, but we could be talking about a different signal because it's fairly stationary for me.   So, a real mystery.

[image]

[image]

jgromes, 2023-04-05:
That's interesting, 125 kHz is obviously not a multiple of the BWs used on SX128x (203.125, 406.25, ...), so probably it is UHF transmitter with an upconverter. I've seen this solution on some satellite communication subsystems before, albeit not for LoRa.

K4KDR, 2023-04-05:
Oh!  Had not thought of that configuration on the TX side.   Thanks for the education!!  Yes, since LoRa S-Band hardware is somewhat new, perhaps these people used existing, reliable LoRa hardware but still wanted to use S-Band as the downlink.   Interesting!