jgromes/RadioLib #721: SX1262 CW signal for calibration

lvplummer, 2023-04-06:
Heltec for their modules using Semtech code has an example program for generating a continuous RF output using the SX1262 module.  It would be very handy if you had a similar tool.

Larry

jgromes, 2023-04-06:
This functionality is already implemented for all modules supported by RadioLib, in fact it's how Morse code, RTTY, SSTV etc. are done. Just call `transmitDirect()` to start transmitting an unmodualted carrier, or `standby()` to stop it.

closed 2023-04-06 by jgromes

lvplummer, 2023-04-06:
Thanks for the information and for the functionality.  I am sorry that I
didn't figure this out on my own.

You have great software which I love to work with plus I really appreciate
your support

Larry