jgromes/RadioLib #722: Module SX1262 is receiving without transmission?

Tintin4000, 2023-04-07:
I am back, as I am stuck again.  I am have now used the following code, mostly a copy-paste of the receive with interrupt example in GitHub.  To my surprise without transmitter the module is constantly receiving data (always the same).  I am still using the Adafruit ESP32 breakout board and use the pin below.

**Sketch that is causing the module fail**

```
#include <Arduino.h>
#include <RadioLib.h>

// Change this to the GPIO connected to the RFM1262 NSS pin
#define RFM1262_NSS 5 // GPIO 33
// Change this to the GPIO connected to the RFM1262 DIO1 pin
#define RFM1262_DIO1 25 // GPIO 14
// Change this to the GPIO connected to the RFM1262 BUSY pin
#define RFM1262_BUSY 36 // GPIO 27
// Change this to the GPIO connected to the RFM1262 RST pin
#define RFM1262_RST 21 // GPIO 32

// ESP32 SPI class
SPIClass *loraSPI = new SPIClass(VSPI);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);

// SX1262 class
//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(RFM1262_NSS, RFM1262_BUSY, RFM1262_RST, RFM1262_DIO1, *loraSPI, spiSettings);

long timeSinceLastPacket = 0;

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// this function is called when a complete packet
// is received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
#if defined(ESP8266) || defined(ESP32)
ICACHE_RAM_ATTR
#endif
void setFlag(void)
{
  // we got a packet, set the flag
  receivedFlag = true;
}

void setup()
{
  // Start serial communication
  Serial.begin(115200);
  Serial.println("SX1262 initializing...");

  // set RF input to high
  pinMode(2, OUTPUT);
  pinMode(17, OUTPUT);
  digitalWrite(2, HIGH);
  digitalWrite(17, HIGH);

  // initialize SPI
  loraSPI->begin(18, 19, 23, 5);

  // Initialize LoRa
  int state = lora.begin(868.0, 125.0, 9, 7);
  if (state != RADIOLIB_ERR_NONE)
  {
    Serial.println("\nLoRa begin failed " + String(state));
  }
  lora.setOutputPower(10);
  lora.setTCXO(1.6);
  // lora.setDio2AsRfSwitch(true);

  // set the function that will be called
  // when new packet is received
  lora.setDio1Action(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1262] Starting to listen ... "));
  state = lora.startReceive();
  if (state == RADIOLIB_ERR_NONE)
  {
    Serial.println(F("success!"));
  }
  else
  {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }
}

void loop()
{
  // check if the flag is set
  if (receivedFlag)
  {
    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = lora.readData(str);

    // you can also read received data as byte array
    /*
      byte byteArr[8];
      int state = radio.readData(byteArr, 8);
    */

    if (state == RADIOLIB_ERR_NONE)
    {
      // packet was successfully received
      Serial.println(F("[SX1262] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1262] Data:\t\t"));
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1262] RSSI:\t\t"));
      Serial.print(lora.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1262] SNR:\t\t"));
      Serial.print(lora.getSNR());
      Serial.println(F(" dB"));
    }
    else if (state == RADIOLIB_ERR_CRC_MISMATCH)
    {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));
    }
    else
    {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);
    }

    // put module back to listen mode
    lora.startReceive();
  }
}
```

Debug:
SX1262 initializing...
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M	SX126x
Symbol length: 1.02 ms
Symbol length: 1.02 ms
Symbol length: 4.10 ms
[SX1262] Starting to listen ... success!
[SX1262] Received packet!
[SX1262] Data:		
[SX1262] RSSI:		-11.50 dBm
[SX1262] SNR:		-2.00 dB

Verbose:
SX1262 initializing...
CMD	80	
DATW	0	FF	
CMD	80	
DATW	0	AA	
CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	1D	3	20	
DATR	0	A2	0	53	0	58	0	31	0	32	0	36	0	31	0	20	0	56	0	32	0	44	0	20	0	32	0	44	0	30	0	32	0	0	
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
M	SX126x
CMD	80	
DATW	0	FF	
CMD	80	
DATW	0	AA	
CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	17	
DATR	0	A2	0	0	0	20	
CMD	C0	
DATR	0	22	0	22	
CMD	7	
DATW	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8F	
DATW	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8A	
DATW	1	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	93	
DATW	20	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	2	
DATW	43	A2	FF	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	89	
DATW	7F	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
Symbol length: 1.02 ms
CMD	8B	
DATW	9	A2	6	A2	3	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
CMD	D	7	40	
DATW	14	A2	24	A2	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
CMD	1D	7	36	
DATR	0	A2	0	D	
CMD	C0	
DATR	0	22	0	22	
CMD	D	7	36	
DATW	D	A2	
CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	D	8	E7	
DATW	18	A2	
CMD	9D	
DATW	1	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	96	
DATW	1	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
Symbol length: 1.02 ms
CMD	8B	
DATW	9	A2	6	A2	3	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
Symbol length: 4.10 ms
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	98	
DATW	D7	A2	DB	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	86	
DATW	36	A2	40	A2	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	1D	8	D8	
DATR	0	A2	0	C8	
CMD	C0	
DATR	0	22	0	22	
CMD	D	8	D8	
DATW	DE	A2	
CMD	1D	8	E7	
DATR	0	A2	0	18	
CMD	C0	
DATR	0	22	0	22	
CMD	95	
DATW	4	A2	7	A2	0	A2	1	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8E	
DATW	A	A2	4	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	D	8	E7	
DATW	18	A2	
CMD	1D	8	E7	
DATR	0	A2	0	18	
CMD	C0	
DATR	0	22	0	22	
CMD	95	
DATW	4	A2	7	A2	0	A2	1	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8E	
DATW	A	A2	4	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	D	8	E7	
DATW	18	A2	
CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	17	
DATR	0	A2	0	0	0	0	
CMD	C0	
DATR	0	22	0	22	
CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	
CMD	C0	
DATR	0	22	0	22	
[SX1262] Starting to listen ... CMD	8	
DATW	2	A2	62	A2	0	A2	2	A2	0	A2	0	A2	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	8F	
DATW	0	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	2	
DATW	43	A2	FF	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	11	
DATR	0	A2	0	1	
CMD	C0	
DATR	0	22	0	22	
CMD	1D	7	36	
DATR	0	A2	0	D	
CMD	C0	
DATR	0	22	0	22	
CMD	D	7	36	
DATW	D	A2	
CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	82	
DATW	FF	A2	FF	A2	FF	A2	
success!
CMD	13	
DATR	0	D2	0	0	0	0	
CMD	C0	
DATR	0	52	0	52	
CMD	80	
DATW	0	D2	
CMD	C0	
DATR	0	22	0	22	
CMD	12	
DATR	0	A2	0	0	0	0	
CMD	C0	
DATR	0	22	0	22	
CMD	13	
DATR	0	A2	0	0	0	0	
CMD	C0	
DATR	0	22	0	22	
CMD	1E	0	
DATR	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	2	
DATW	43	A2	FF	A2	
CMD	C0	
DATR	0	22	0	22	
[SX1262] Received packet!
[SX1262] Data:		
[SX1262] RSSI:		CMD	14	
DATR	0	A2	0	6E	0	F8	0	17

jgromes, 2023-04-07:
This pinout seems rather strange:

```c++
SX1262 lora = new Module(RFM1262_NSS, RFM1262_BUSY, RFM1262_RST, RFM1262_DIO1, *loraSPI, spiSettings);
```

The pin order is NSS, DIO1, NRST, BUSY, so your BUSY and DIO1 pins seem swapped.

Tintin4000, 2023-04-07:
OMG!  You are spot on.  Did try to swap on the receiver and send data from the transmitter with success.  I must have copy-paste this from somewhere by mistake.  Once again amazing quick response.  Thank you so much.

closed 2023-04-07 by Tintin4000