jgromes/RadioLib #724: sx128x invertIQ functional?

K4KDR, 2023-04-08:
Greetings!

With sx127x devices, I see in 

`src/modules/SX127x/SX127x.h`

reference to the directive '**invertIQ**'

`int16_t invertIQ(bool invertIQ);`

which I am able to utilize in sketches successfully to set (or not set) the device to invert the received signals

[image]

... however, I don't understand the programming syntax well enough to understand how to accomplish the same thing with sx128x devices.

In

`src/modules/SX128x/SX128x.h`

... I see references to inverted -vs- non-inverted

[image]

... but I apologize in advance for not being able to figure out how to instruct the hardware to use one setting or the other.

When you have a rare spare moment, could I please get some direction on how to do that with **sx128x devices**?   Searching through the technical documentation & generic searches online have not given me anything to help.

Many thanks as always!

-Scott,  K4KDR

jgromes, 2023-04-08:
Looks like IQ inversion was never implemented for SX128x, and for SX126x for that matter. I added both, though it's untested, so use with caution.

closed 2023-04-08 by jgromes

K4KDR, 2023-04-08:
Thanks SO much!!

I'll try known signals and verify that it works one way & NOT the other.