jgromes/RadioLib #727: Heltec v3 using SX1262 losing packets

klemzi, 2023-04-11:
Hi @jgromes, I am having similar issue with this using this library. We were using the heltec v2 which is using the SX1276 chip and we had no issues. Now, it is phased out, so we decided to try out the new v3 (uses SX1262) using RadioLib. However trying out the example code for transmitting and receiving, found out that the receiver node is dropping or losing a lot of packets.
I am sending about 68 bytes of data. to be able to do this to reduce the packet loss, I transmit every 1s and then set the SF to 10 and coding rate to 5. but if i go above this bytes or send let's say every 0.5s then there will be lots of packet loss. Here's the receiver output so far:

[image]

I was going to create a new issue, but this looks similar to mine. I need help, thanks! I am trying out the with interrupt example next.

jgromes, 2023-04-11:
As per my original comment: 

> Please open a new issue (and maybe try the Receive_Interrupt examples instead of the blocking ones) with all the information requested by the appropriate template.

klemzi, 2023-04-11:
Yes. I am trying out the non blocking ones at the moment.

klemzi, 2023-04-12:
Worked! thanks @jgromes. Went ahead to try it with radioHead as well and no packets drop. sent 250 packets during this test and got 250 at the receiver with no retry data. Thank you!!! I can close this now.

[image]

closed 2023-04-12 by klemzi