jgromes/RadioLib #731: SX1262 + invertIQ()

RonMeier, 2023-04-16:
I use your library on one with an SX1262 (SX1262 radio = new Module()) (heltec esp32 V3)

unfortunately invertIQ() does not work.
if I set invertIQ(true) the IQ is never set to true.

 - Arduino IDE version 1.8.19
 - Library version 5.7.0

jgromes, 2023-04-17:
> if I set invertIQ(true) the IQ is never set to true.

I don't understand what exactly do you mean, could you elaborate?

jgromes, 2023-04-17:
Initially, I was not able to replicate this. Digging through the code, I did finally find a problem with the IQ inversion, but I have to say your description did not help at all.

> It is not possible to receive packets with inverted IQ or to send them with inverted IQ set.

From this report, I have no idea what you tried, what you expected to happen and what really happened. Does the receiver timeout indefinitely when invertIQ is set to true? Are packets no longer being transmitted? What code and/or example did you try? From your description (or rather lack thereof) it's impossible to tell.

> It seems to me that invertIQ cannot be set to "true".

Again, what was the problem you were facing? That the value `true` cannot be passed to `invertIQ`? Probably not, but it's impossible to know from your description.

So next time, please use the issue templates. For some reason, I have to repeat this on nearly every other issue, which is taking my time as well as yours.

closed 2023-04-17 by jgromes