jgromes/RadioLib #733: CC1101 + ESP32 receiving random junk data

omaryangtw, 2023-04-20:
Hi,

I setup 2 ESP32 + CC1101, with 1 transmitting and 1 receiving.

Things are fine if I use [CC1101_Receive.ino](https://github.com/jgromes/RadioLib/blob/master/examples/CC1101/CC1101_Receive/CC1101_Receive.ino) code. 
Following is the serial output:

[CC1101] Data:		Hello World!Hello World!Hello World!Hello World!
[CC1101] RSSI:		-15.50 dBm
[CC1101] LQI:		12

But when I use [CC1101_Receive_Interrupt.ino](https://github.com/jgromes/RadioLib/blob/master/examples/CC1101/CC1101_Receive_Interrupt/CC1101_Receive_Interrupt.ino)
The message output will be like:

[CC1101] Received packet!
[CC1101] Data:		P�M�,�J����8�H$�Tפߙ��{�w�A%�l��H�q-}��

[CC1101] RSSI:		-120.00 dBm
[CC1101] LQI:		95

Or just a sequence of 
CRC error!

And the message differ every time I reboot the ESP32.

**Additional info:**
 - MCU: ESP-WROOM-32
 - Wireless module: CC1101
 - Arduino IDE version: 2.0.4  
 - Library version: 5.7.0

jgromes, 2023-04-20:
Possibly related to #345, I will try to investigate this.

lewisxhe, 2023-07-07:
I encountered the same problem. After rolling back the 5.1.0 version, the garbage data is reduced. Using the master branch, the reception interruption will continue to come and cannot be stopped

jgromes, 2023-07-08:
I was able to replicate this issue but so far no luck fixing it. I think the root cause is the interrupt keeps firing, possibly because RX FIFO is not being flushed. So gold star to anyone who can figure this out before me ;)

jgromes, 2023-08-02:
I was able to resolve this, the root cause was that the interrupt direction was wrong (rising instead of falling). However, there seemed to have been additional problems with the continuous FIFO writing/reading to facilitiate packets longer than 64 bytes. For the time being I dropped this feature from CC1101, since I'm already working on implementing Stream mode for CC1101, which will allow arbitrarily large packets, same as for SX127x and other modules.

closed 2023-08-02 by jgromes