jgromes/RadioLib #734: E22-900M22S not going to sleep

Tintin4000, 2023-04-21:
I have an E22-900M22S ebyte module connected to an ESP32 board.  Upon issuing the sleep command from a standby state, the module consumption stays the same.  It is around 545µA, which is consistent for the standby mode, but do not drop any further down after the sleep command is issued.  I have tried multiple ebyte modules and also 2 different ESP32 boards, same same.
The SPI bus stays active during the life cycle.
I will activate the debug mode to see if there is anything abnormal and post the result here.
Any idea what could be the cause?

jgromes, 2023-04-22:
First, please post your code **as per the issue template ...**

Next, `sleep()` by default uses warm start, is there a difference for cold start using `sleep(false)`? Also, where and how are you measuring the power consumption?

Tintin4000, 2023-04-22:
The code is really simple, receiving or transmitting result in the same observation.
I tried warm and cold sleep no difference.

I am measuring the power consumption of the module on the VCC line with a Power Profiler Kit, multimeter give the same results.

The 2 command Standby and Sleep are debugged here under.  Question, why is there a GetStatus request after each?  The datasheet said that after issuing the SetSleep command, the module should not receive any further command, except for waking it up.

CMD	80	
DATW	0	A2	
CMD	C0	
DATR	0	22	0	22	
CMD	84	
DATW	4	A2	
CMD	C0	
DATR	0	22	0	22	

```
#include <Arduino.h>
#include <RadioLib.h>

// Change this to the GPIO connected to the SX1262 NSS pin
#define SX1262_NSS 5
// Change this to the GPIO connected to the SX1262 DIO1 pin
#define SX1262_DIO1 25
// Change this to the GPIO connected to the SX1262 BUSY pin
#define SX1262_BUSY 36
// Change this to the GPIO connected to the SX1262 RST pin
#define SX1262_RST 21

// ESP32 SPI class
SPIClass *loraSPI = new SPIClass(VSPI);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);

// SX1262 class
SX1262 lora = new Module(SX1262_NSS, SX1262_DIO1, SX1262_RST, SX1262_BUSY, *loraSPI, spiSettings);

void setup()
{
  // Start serial communication
  Serial.begin(115200);

  Serial.println("SX1262 initializing...");

  // initialize SPI
  loraSPI->begin(18, 19, 23, 5);

  // Initialize LoRa
  int state = lora.begin(868.0, 125.0, 9, 7);
  if (state != RADIOLIB_ERR_NONE)
  {
    Serial.println("\nLoRa begin failed " + String(state));
  }
  lora.setOutputPower(10);
  lora.setTCXO(1.6);
  lora.setDio2AsRfSwitch(true);

  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  state = lora.transmit("Hello");

  if (state == RADIOLIB_ERR_NONE)
  {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(lora.getDataRate());
    Serial.println(F(" bps"));
  }
  else if (state == RADIOLIB_ERR_PACKET_TOO_LONG)
  {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));
  }
  else if (state == RADIOLIB_ERR_TX_TIMEOUT)
  {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));
  }
  else
  {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }
  // set module to sleep
  lora.standby();
  lora.sleep();

}

void loop()
{
}
```

jgromes, 2023-04-22:
OK, I think I know what's going on here - you probably have the latest library version (5.7.0) from Arduino library manager or Platformio correct? Though I have to say the issue template does ask for this information, so had you followed it you would have had the answer sooner.

This has actually been already resolved in #705. However, there wasn't yet a new release, so unless you use the latest master, you will have to wait a bit for the next one.

The good news is that a new release is coming soon, in the next couple of days.

closed 2023-04-22 by jgromes

Tintin4000, 2023-04-22:
Very sorry for that, I will be thorough in the future.  Yes the version I have is 5.7.0.  However following the #705, the solution seems to set verify to false which if I am correct is already the case in the sleep implementation for the SX126x?  See below.

```
int16_t` SX126x::sleep(bool retainConfig) {
  // set RF switch (if present)
  _mod->setRfSwitchState(Module::MODE_IDLE);

  uint8_t sleepMode = RADIOLIB_SX126X_SLEEP_START_WARM | RADIOLIB_SX126X_SLEEP_RTC_OFF;
  if(!retainConfig) {
    sleepMode = RADIOLIB_SX126X_SLEEP_START_COLD | RADIOLIB_SX126X_SLEEP_RTC_OFF;
  }
  int16_t state = _mod->SPIwriteStream(RADIOLIB_SX126X_CMD_SET_SLEEP, &sleepMode, 1, false);

  // wait for SX126x to safely enter sleep mode
  _mod->delay(1);

  return(state);
}```

jgromes, 2023-04-22:
That's correct, however, the first `false` just disables waiting for the BUSY signal. This is the current implementation:

https://github.com/jgromes/RadioLib/blob/767a2b006a0349c683624b33d3be3e8a9d5330c3/src/modules/SX126x/SX126x.cpp#L432-L446

Tintin4000, 2023-04-23:
I have downloaded the latest code from the master branch and I can confirm that the implementation change is working for the ebyte module.  Going to sleep and consuming around 800nA. Perfect, thank you again for your help.