jgromes/RadioLib #735: SX1262 Sleep

lvplummer, 2023-04-21:
Using latest RadioLib release.  

My setup relative to this issue is pretty simple and not changed from earlier releases that work correctly.  Radio module is a NiceRF SX1262 for 915 MHz driven by a minimalist Uno - chip, crystal, and two caps.  RTC causes once per minute interrupts where SX1262 transmits packet, receives updates, and sleeps.  

First cycle prior to first sleep works correctly.  Future cycles after SX1262 sleeps do work correctly.  No transmit and blocked for 15 seconds.

Work around is to reset (begin) SX1262 before transmitting but that adds 250 milliseconds using too much battery.  Another work around is to use standby instead of sleep and that increases current which also uses too much battery.

Larry

jgromes, 2023-04-22:
First, please post your code and debug mode output **as per the issue template ...**

> First cycle prior to first sleep works correctly. Future cycles after SX1262 sleeps do work correctly.

Bit confused here. First works and future works as well ...? I'm guessing you meant "do not work correctly". If that is the case, then please define what does "do not work correctly" mean. "No transmit and blocked for 15 seconds." is not nearly enough, post the serial/debug mode output.

lvplummer, 2023-04-22:
You are correct, I meant "does not work correctly."

I don't have time right now to run the debug mode but I have attached my
sketch.  In addition, essentially the same sketch for the SX1276 sleeps
correctly.

jgromes, 2023-04-22:
Files you attach to the reply email will not get posted to the issue.

Also, see the resolution of #734, seems somewhat similar to your problem. If not, then I'm afraid not much can be done without the debug output.

lvplummer, 2023-04-22:
I am running the latest release from GitHub.  Sounds like there will be a
new release soon.  I will first check to see if that resolves my issue.

lvplummer, 2023-04-24:
SX1262 sleep in the version released on March 27 works correctly.  For all subsequent releases, sleep does not work correctly.  It seems that the radio configuration is not preserved.

jgromes, 2023-04-25:
> SX1262 sleep in the version released on March 27

There was no release on March 27. There was 5.7.0 on March 5th. There were a couple of commits on March 27th, but since then nothing obviously related to sleep functionality was done.

So at the risk of repeating myself, unless you can provide the information asked for by the issue template (library version, sketch you are using and debug mode output), there's not much that can be done.

lvplummer, 2023-04-25:
Sorry that I used the wrong term.  I am talking about the commits on those
dates.

I don't understand what is happening in your software.  I can only
report the results.

I will continue to use earlier commits that function correctly.

closed 2023-04-25 by jgromes

lvplummer, 2023-04-25:
> "My software" does not work in isolation - how it's used matters, a lot.
> Debug is there to help in these exact situations, but if you don't provide
> that information, I don't know how exactly am I supposed to help, sorry.
> Closing until further information is provided.

OK.