jgromes/RadioLib #738: RFM69HC with ESP8266-12F intterupt on DIO0 receiving  example not working.

ramanc51, 2023-04-28:
Hello jgromes thanks to your good effort on development of such a library to support multiple RF chips

i am using RFM69HC-434MHz  with ESP8266-12F, running interrupt DIO0 example, Serial output given bellow, no data receiver side   how to debug this issue?

Transmitter side
[SX1231] Initializing ... success!
chiVersion: 36
[RF69] Setting high power module ... success!
[RF69] Sending first packet ... transmission finished!
[RF69] Sending another packet ... transmission finished! 

Recevier side
[SX1231] Initializing ... success!
chipRevision: 36
[RF69] Setting high power module ... success!
[RF69] Starting to listen ... success!

No more data received at receiver side, how to debug sir

Receivi

ramanc51, 2023-04-29:
I enabled debug mode,  i got following info,  same info on both transmitter and receiver board

[SX1231] Initializing ... RF69 not found! (1 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024

PIN Connection

ESP8266     RFM69HC - 434MHz
MOSI          GPIO13
MISO          GPIO12
CLK             GPIO14
CS               GPIO15
NSS             GPIO4
DIO0            GPIO5   

If i change following 

SPI1U  = SPIUMOSI | SPIUDUPLEX | SPIUSSE;

on  SPIClass::begin()  I got following 

Recevier side
[SX1231] Initializing ... success!
chipRevision: 36
[RF69] Setting high power module ... success! 

I think SPI setting may cause issue, any idea to resolve it?

jgromes, 2023-04-29:
Please post the actual code you are running, and the library version. Like I said previously, `chipRevision: 36` is not in this version of the library, so yours is either outdated, or altered.

ramanc51, 2023-04-30:
Libray version used: 6.0.0

Receiver code
/*
   RadioLib SX1231 Receive Example

   This example receives packets using SX1231 FSK radio module.

   NOTE: SX1231 offers the same features as RF69 and has the same
         interface. Please see RF69 examples for examples on AES,
         address filtering, interrupts and settings.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#rf69sx1231

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
// SX1231 has the following connections:
// CS pin:    10
// DIO0 pin:  2
// RESET pin: 3
RF69 radio = new Module(15, 5, 4, 0xff);
// flag to indicate that a packet was received
volatile bool receivedFlag = false;      
// or using RadioShield
// https://github.com/jgromes/RadioShield

//SX1231 radio = RadioShield.ModuleA;
IRAM_ATTR void setFlag(void) {

        //we got a packet, set the flag
        receivedFlag = true;  
}

void preHwInit(void){

        // Reset the RFM69 Radio
        pinMode(4, OUTPUT);
        digitalWrite(4, HIGH);
        delayMicroseconds(100);
        digitalWrite(4, LOW);
        delay(10);
}

void setup() {

        delay(100);
        ESP.wdtDisable();
        Serial.begin(9600);
        delay(100);preHwInit();

        // initialize SX1231 with default settings
        Serial.print(F("[SX1231] Initializing ... "));

        int state = radio.begin();
        if (state == RADIOLIB_ERR_NONE) {
          Serial.println(F("success!"));
          Serial.print(F("chipVersion: "));
          Serial.println(radio.getChipVersion());          
        } else {
          Serial.print(F("failed, code "));
          Serial.println(state);
          while (true);
        }

        //set the function that will be called when new packet is received
[... 157 lines cut ...]
        ESP.wdtEnable(WDTO_8S); yield();
}

void loop() {

            // check if the previous transmission finished
        if(transmittedFlag) {
         
          // reset flag
          transmittedFlag = false;

          if (transmissionState == RADIOLIB_ERR_NONE) {
            // packet was successfully sent
            Serial.println(F("transmission finished!"));   
          } else {
            Serial.print(F("failed, code ")); Serial.println(transmissionState);
          }

          radio.finishTransmit(); delay(1000);
          // send another one
          Serial.print(F("[RF69] Sending another packet ... "));
          // you can transmit C-string or Arduino string up to 256 characters long
          transmissionState = radio.startTransmit("Hello World!");
        }       
}

jgromes, 2023-04-30:
OK, thanks. Couple of observations:

1. On the transmitter, you are using the SX1231 class with RF69 hardware. Why is that? Or do oyu have a differnet hardware for the transmitter?
2. Your `preHwInit` function resets the RF69 module. This is already handled by the library, there's no need to do that manually.
3. Please use `RADIOLIB_NC` instead of the `0xff` for unused pins.

Also, your pinout is very strange, you claimed to have the following:

CS - GPIO15
NSS - GPIO4

CS is "chip select", also sometimes called "negated slave selected", or NSS. These are the same pin, yet you claim to have connected to different pins ... ? There should be just one NSS or CS pin on your RF69.

ramanc51, 2023-04-30:
Sorry mistake on my pin labling, correct pin connection is given below.

RFM69 - ESP8266-12F

MOSI - GPIO13
MISO - GPIO12
CLK   - GPIO14
NSS  - GPIO15
RST  -  GPIO4
DIO0 - GPIO5

ramanc51, 2023-05-01:
now i initialized on both TX/RX  with RF69, commented preHwInit, now it is working, on receiver side i got following

[RF69] Received packet!
[RF69] Data:            Hello World!
[RF69] RSSI:            -44.00 dBm
[RF69] Received packet!
[RF69] Data:            Hello World!
[RF69] RSSI:            -48.50 dBm
[RF69] Received packet!
[RF69] Data:            Hello World!
[RF69] RSSI:            -44.50 dBm

jgromes, 2023-05-01:
I'm glad you were able to get it working!

However, I would advise against modyfing SPI.cpp. Instead, you should find out why you need to modify this, what exactly are the flags doing and how to it from the sketch/the public API of the SPI library.

closed 2023-05-01 by jgromes