jgromes/RadioLib #740: SX1262 Sleep

lvplummer, 2023-05-02:
My board is very simple - Arduino UNO chip and crystal with no regulator or serial interrace - battery powered.  I made two changes to your transmit example without interrupts:  pin assignments to match my board and serial baud rate changed to 115200.

I turned on debug and compiled your blocking SX1262 transmit example.  Here are the results from that:

18:03:36.088 -> [SX1262] Initializing ... Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
18:03:36.088 -> 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
18:03:36.088 -> 
18:03:36.088 -> M	SX126x
18:03:36.088 -> Symbol length: 4719 ms
18:03:36.136 -> Symbol length: 4719 ms
18:03:36.136 -> Symbol length: 4719 ms
18:03:36.136 -> success!
18:03:36.136 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:03:36.278 -> success!
18:03:36.278 -> [SX1262] Datarate:	567.64 bps
18:03:37.273 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:03:37.462 -> success!
18:03:37.462 -> [SX1262] Datarate:	567.64 bps
18:03:38.455 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:03:38.643 -> success!
18:03:38.643 -> [SX1262] Datarate:	567.64 bps
18:03:39.633 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:03:39.822 -> success!
18:03:39.822 -> [SX1262] Datarate:	567.64 bps
18:03:40.811 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:03:40.999 -> success!

I then inserted a radio.sleep() just before the one second delay.  

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  radio.sleep();
  // wait for a second before transmitting again
  delay(1000);
}

Here are the results of that:

18:09:34.496 -> [SX1262] Initializing ... Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
18:09:34.496 -> 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
18:09:34.496 -> 
18:09:34.496 -> M	SX126x
18:09:34.496 -> Symbol length: 4719 ms
18:09:34.544 -> Symbol length: 4719 ms
18:09:34.544 -> Symbol length: 4719 ms
18:09:34.544 -> success!
18:09:34.544 -> [SX1262] Transmitting packet ... Timeout in -8704 us
18:09:34.686 -> success!
18:09:34.686 -> [SX1262] Datarate:	567.70 bps
18:09:35.711 -> [SX1262] Transmitting packet ... Timed out waiting for GPIO pin, is it connected?
18:09:36.714 -> failed, code -705
18:09:37.708 -> [SX1262] Transmitting packet ... Timed out waiting for GPIO pin, is it connected?
18:09:38.694 -> failed, code -705
18:09:39.723 -> [SX1262] Transmitting packet ... Timed out waiting for GPIO pin, is it connected?
18:09:40.703 -> failed, code -705

The first transmission before the first sleep works correctly.   Subsequent transmission fail.

jgromes, 2023-05-02:
Thanks for the report, I was able to find this one quickly. There was a bug in waking up the SX126x device after it was put to sleep (the Arduino just timed out waiting for the BUSY signal to go low, which is not going to happen when the device is sleeping). I added a check to make sure that it will wakeup. Tested on Uno + SX1268, thanks for reporting!

closed 2023-05-02 by jgromes

lvplummer, 2023-05-02:
Thanks for your great software and also the quick support.

However, your solution seems to have added one second somewhere.  My
application requires long battery life which means that CPU and
radio current during sleep must be very low AND a transmit cycle needs to
be very short.

Your sleep fix has caused my transmit cycle to go from 190 milliseconds to
1.2 seconds.

lvplummer, 2023-05-02:
Seems that the additional second is not coming from sleep but rather from
transmit which I assume does the wakeup.

jgromes, 2023-05-02:
I see - that's caused by the timeout while waiting for BUSY, which is one second.

I added additional parameter to the `standby()` method, so if the one second waiting time is critical for you, you can wakeup the module immediately like this:

```c++
radio.sleep();
// radio sleeping here
(...)

// OK, time to wake up
radio.standby(RADIOLIB_SX126X_STANDBY_RC, true);

// radio in standby, transmit
radio.transmit(...)
```

lvplummer, 2023-05-02:
Wow!  Your real time support is amazing.

I just tried your solution and it works.

I hope not to have to bug you for a while.

Larry

lvplummer, 2023-05-02:
Sorry to still be a pain.

I have three simple battery powered remote telemetry gadgets that send data
one per minute.  I tried your fix on all three.  Two work correctly and one
does not.

I ran your modified example sketch with the standby fix on all three in
debug mode.  Here are the results from one that works and one that
doesn't.  I tried switching the UNO chip and the issue is only related to
the SX1262 module.  I am guessing some timing parameter is right on the edge

A reset (begin)

14:01:48.846 -> [SX1262] Initializing ... Found SX126x:
RADIOLIB_SX126X_REG_VERSION_STRING:
14:01:48.846 -> 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 |
SX1261 V2D 2D02.
14:01:48.846 ->
14:01:48.846 -> M SX126x
14:01:48.846 -> Symbol length: 4719 ms
14:01:48.846 -> Symbol length: 4719 ms
14:01:48.846 -> Symbol length: 4719 ms
14:01:48.846 -> success!
14:01:48.846 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:01:49.035 -> success!
14:01:49.035 -> [SX1262] Datarate: 567.64 bps
14:01:50.027 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:01:50.216 -> success!
14:01:50.216 -> [SX1262] Datarate: 567.64 bps
14:01:51.210 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:01:51.387 -> success!

---------------------------------------------------

14:04:03.425 -> [SX1262] Initializing ... Found SX126x:
RADIOLIB_SX126X_REG_VERSION_STRING:
14:04:03.425 -> 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 |
SX1261 V2D 2D02.
14:04:03.425 ->
14:04:03.425 -> M SX126x
14:04:03.425 -> Symbol length: 4719 ms
14:04:03.425 -> Symbol length: 4719 ms
14:04:03.425 -> Symbol length: 4719 ms
14:04:03.425 -> success!
14:04:03.425 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:04:03.661 -> timeout!
14:04:04.648 -> [SX1262] Transmitting packet ... failed, code -2
14:04:05.647 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:04:05.836 -> success!
14:04:05.836 -> [SX1262] Datarate: 550.33 bps
14:04:06.821 -> [SX1262] Transmitting packet ... failed, code -2
14:04:07.855 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:04:07.994 -> success!
14:04:07.994 -> [SX1262] Datarate: 550.29 bps
14:04:09.042 -> [SX1262] Transmitting packet ... failed, code -2
14:04:10.011 -> [SX1262] Transmitting packet ... Timeout in -8704 us
14:04:10.197 -> success!
14:04:10.197 -> [SX1262] Datarate: 550.31 bps
14:04:11.201 -> [SX1262] Transmitting packet ... failed, code -2

jgromes, 2023-05-03:
That's interesting, couple of thoughts:

1. Is there any difference between the SX126x module that works and the others? I'm guessing  the rest of the hardware (the MCU, power etc.) is the same for all three.
2. Could you run the module that does not work with both DEBUG and VERBOSE enabled?

lvplummer, 2023-05-03:
1.  I have a total of five transmitter boards - four in one band and one in
another.  All are identical and very simple - UNO DIP chip, crystal, two
caps, DS3231 RTC, and two AAA batteries.  The CPU is in a DIP socket so I
swapped the CPU chip and the issue aligned with the radio module.  The
modules for the two bands are from two different manufacturers.  NiceRF for
the four 915 MHz modu;es and another Chinese company that I can't remember
the name of for the one 433 MHz band.

2.  I tried turning on both DEBUG and VERBOSE where the output display was
the same that I sent before.

Even though the unit with the" bad" module works except when I use sleep, I
think it is time to call this module defective and move on.  If I see
something similar from the boards that are known good, I will let you know.

Larry

jgromes, 2023-05-03:
OK, let me know if the issue pops up again, I'd like to get to the bottom of this.

lvplummer, 2023-05-03:
I am up for helping anyway I can.  Let me know if you have ideas or code to
test.

lvplummer, 2023-05-03:
With no sleep and DEBUG and VERBOSE on, here is the output of one cycle:

10:48:19.232 -> [SX1262] Initializing ... CMD 80
10:48:19.515 -> DATW 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 1D 3 20
10:48:19.515 -> DATR 0 A2 0 53 0 58 0 31 0 32 0 36 0 31 0 20 0 56 0 32 0 44
0 20 0 32 0 44 0 30 0 32 0 0
10:48:19.515 ->
10:48:19.515 -> Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
10:48:19.515 -> 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 |
SX1261 V2D 2D02.
10:48:19.515 ->
10:48:19.515 -> M SX126x
10:48:19.515 -> CMD 80
10:48:19.515 -> DATW 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 80
10:48:19.515 -> DATW 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 80
10:48:19.515 -> DATW 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 17
10:48:19.515 -> DATR 0 A2 0 0 0 0
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 97
10:48:19.515 -> DATW 0 A2 0 A2 1 A2 40 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 8F
10:48:19.515 -> DATW 0 A2 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 8A
10:48:19.515 -> DATW 1 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 93
10:48:19.515 -> DATW 20 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD 88
10:48:19.515 -> DATW 3 A2 16 A2 A A2 0 A2 0 A2 0 A2 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
[... 247 lines cut ...]
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.515 -> CMD D 8 89
10:48:19.515 -> DATW 4 A2
10:48:19.515 ->
10:48:19.515 -> CMD 83
10:48:19.515 -> DATW 0 A2 0 A2 0 A2
10:48:19.515 ->
10:48:19.515 -> CMD C0
10:48:19.515 -> DATR 0 22 0 22
10:48:19.515 ->
10:48:19.752 -> CMD 2
10:48:19.752 -> DATW 43 A2 FF A2
10:48:19.752 ->
10:48:19.752 -> CMD C0
10:48:19.752 -> DATR 0 22 0 22
10:48:19.752 ->
10:48:19.752 -> CMD 80
10:48:19.752 -> DATW 0 A2
10:48:19.752 ->
10:48:19.752 -> CMD C0
10:48:19.752 -> DATR 0 22 0 22
10:48:19.752 ->
10:48:19.752 -> timeout!

lvplummer, 2023-05-04:
> Please disregard the previous info.  Sleep was off while standby was on.
>
> Here are results from sleep and standby on with no debug or verbose.
> Seems to fail on alternate attempts.
>
> 11:38:42.478 -> [SX1262] Initializing ... success!
> 11:38:42.478 -> [SX1262] Transmitting packet ... timeout!
> 11:38:52.777 -> [SX1262] Transmitting packet ... failed, code -2
> 11:39:02.756 -> [SX1262] Transmitting packet ... success!
> 11:39:02.931 -> [SX1262] Datarate: 550.19 bps
> 11:39:12.932 -> [SX1262] Transmitting packet ... failed, code -2
> 11:39:22.934 -> [SX1262] Transmitting packet ... success!
> 11:39:23.111 -> [SX1262] Datarate: 550.17 bps
> 11:39:33.093 -> [SX1262] Transmitting packet ... failed, code -2
> 11:39:43.115 -> [SX1262] Transmitting packet ... success!
> 11:39:43.305 -> [SX1262] Datarate: 550.18 bps
> 11:39:53.286 -> [SX1262] Transmitting packet ... failed, code -2
> 11:40:03.295 -> [SX1262] Transmitting packet ... success!
> 11:40:03.484 -> [SX1262] Datarate: 550.17 bps
> 11:40:13.465 -> [SX1262] Transmitting packet ... failed, code -2
>
> With DEBUG and VERBOSE on (sleep and standby still on), seems to always
> work.  I am guessing that debug changes timing.
>
> 11:43:20.802 -> [SX1262] Initializing ... CMD 80
> [... 556 lines cut ...]
> 11:43:41.484 -> success!
> 11:43:41.484 -> [SX1262] Datarate: 564.99 bps
> 11:43:41.484 -> CMD 84
> 11:43:41.484 -> DATW 4 A2
> 11:43:41.484 ->

I mentioned earlier that my transmitter is very simple so that I can
minimize current draw during sleep - UNO DIP chip, crystal, 2 caps, SX1262
module, and 2 AAA batteries.  I tried different UNO chips to make sure the
issue was not the CPU.  That caused me to focus on the radio module.

By doing some timing measurements using the CPU millis() timer versus
external oscilloscope, I determined that the CPU was running at about 4 MHz
instead of 16 MHz.

I replaced the bad crystal and all boards with radios from two sources
running on two bands are working as expected.

jgromes, 2023-05-04:
> the CPU was running at about 4 MHz instead of 16 MHz

Thanks for letting me know - to be honest, I'm rather surprised it was working at all at that frequency, because the default SPI frequency is 2 MHz.

lvplummer, 2023-05-04:
I may not have been correct with the 4 MHz statement.  It may have been
intermittent.   I am glad that one is resolved.