jgromes/RadioLib #743: Be able to send multiple pocsag message at once.

Bastiti, 2023-05-09:
Hello, 

I'm currently building a pocsag transmitter connected to internet and the issue is that I cannot send multiple message at once.

I have 6.4s on a timeslot where I can transmit and I do not want it to replicate the header which cause a massive time loss.

so my request is to add a function that would allow me to send multiple message to multiple pager.

Thanks for your answer

Bastien ON4BCY

jgromes, 2023-05-09:
Sending multiple messages in one batch is a feature of the POCSAG protocol and you're right, at the moment it is not implemented in RadioLib. The reason is that supporting this somewhat complicates implementation, and has some downsides. For example, whether your messages will be sent in one batch (in POCSAG terminology) will depend on their size and destination address.

Nothing is stopping this from being added, I just didn't think there was that much benefit to it honestly. I'm open to being convinced otherwise of course :)

Bastiti, 2023-05-10:
Ham Radio operator use DAPNET.
DAPNET stands for Decentralized Amateur Paging Network. It is an open-source network that allows amateur radio operators to send short messages via pocsag.
DAPNET can be useful to communicate for emergency situations etc.

There is a high amount of user in Germany, soon in Belgium, ...

We can receive up to 50messages to transmit. But we're limited by timeslots of 6.4seconds.

And at the moment, I might be the only one using dapnet on a ESP32. But I'm quite stuck with RadioLib as I loose so much time with the header. It would be a cheap DAPNET Transmitter.

Bastiti, 2023-05-10:
And I'm not confortable enough in C/C++ to add that part.

jgromes, 2023-05-10:
DAPNET was one the original use case for RadioLib support of POCSAG (when it got first requested some 4 years ago). I never used it myself, though it seems interesting.

I'll see if this can be implemented. Of course, pull requests are welcome, from any member of the DAPNET community.