jgromes/RadioLib #747: delay(), LOW and HIGH undeclared on CC1101 on ESP-IDF

nicklasb, 2023-05-12:
**Describe the bug**
I CC1101.cpp, LOW and HIGH are assumed to be defined. They are not on ESP-IDF (and perhaps other platforms). 
Also on line 135, delay(20) is called, not this->mod->hal->delay(20), which also causes compiler errors. 

**To Reproduce**
Compile Radiolib on some none-arduino platform. without Arduino.h (seems to be Common.h actually).

**Expected behavior**
No compiler errors, that the HAL is to be used in all cases.
**Additional info (please complete):**
 - MCU: ESP32 of different kinds. (and possibly also on pure STM32 cube and others without Arduino-stuff)
 - Wireless module type: Actually it is SX1262 and SX1278, but CC1101.cpp seems to be included in compile anyway, probably by ESP-IDF. 
 - Arduino IDE version  N/A
 - Library version ^6.0.0 (current PlatformIO release)

jgromes, 2023-05-13:
This was fixed in 3df9cb27ff55fcb34fde6ee9ccff4bccc3426d2a, but there was no release yet, so you won't see it in the platformio package. As a workaround, you can use the latest master instead of the 6.0.0, or just wait until the next update.

closed 2023-05-13 by jgromes

nicklasb, 2023-05-13:
> This was fixed in [3df9cb2](https://github.com/jgromes/RadioLib/commit/3df9cb27ff55fcb34fde6ee9ccff4bccc3426d2a), but there was no release yet, so you won't see it in the platformio package. As a workaround, you can use the latest master instead of the 6.0.0, or just wait until the next update.

Great! 
I didn't find that when I searched (looked for the other keywords..)

Thanks!