jgromes/RadioLib #748: Cannot load RadioLib.h and can

nicklasb, 2023-05-14:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please use Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

nicklasb, 2023-05-15:
Just to be clear, realized that I had not actually said it, but for indexing: 
`#include <RadioLib.h>` does not work after following:
https://registry.platformio.org/libraries/jgromes/RadioLib/installation

If I use a *relative* path (../../../.pio/XX/libdeps and so on), it works, but then the relative path will fail when other parts of the framework wants to include it. (seems that relative paths aren't all that relative).

nicklasb, 2023-05-15:
I changed to depend on https://github.com/jgromes/RadioLib.git instead and still it didn't work. 
However, that had a CMakeLists.txt, which should've helped.
Annoyingly, only VSCode resolves it, not PlatformIO (or ESP-IDF).

nicklasb, 2023-05-15:
Ok. Well. This seems all to be about the compiler being confused with C/C++. Or atleast that is my theory.

After putting all the C++ stuff, like the RadioLib classes in one file, so that RadioLib did not have to be imported by anyone else, it started compiled again.  

However, and I think that this might be part of the issue; I still cannot include RadioLib normally.
```
#include <RadioLib.h>
#include <RadioLib/RadioLib.h>
#include <RadioLib/src/RadioLib.h>

```
doesn't work. 

Only thing that works is:
`#include "/full_path_to_project/RobArch/.pio/libdeps/TTGO-LoRa32-v1/RadioLib/src/RadioLib.h"`

Regardless if I am using the latest release or master.

jgromes, 2023-05-16:
@nicklasb in the last few commits, the dependency on algorithm should be resolved. As to the rest of your issues, I'll try to replicate them with ESP-IDF. If you could share your HAL implementation and Cmake, that would help me get started.

Mesteery, 2023-05-16:
I don't think defining min and max as macro is a good solution. It will alter with user code, and will prevent him from having e.g variable/functikn named max or min..

nicklasb, 2023-05-17:
@jgromes Hi, it doesn't seem like the issue was related to the <algorithm> dependency. 
It doesn't include after that fix either, and I have not yet succeeded in replicating it in a smaller project. 
I created a small project (https://github.com/nicklasb/RadioLib_PIO_ESPIDF) with an empty HAL and there it compiles. 
I think that you might be on to something with that it has something to do with the HAL. 
I will keep trying tomorrow, if I can make this work it'd be great.

nicklasb, 2023-05-17:
Hi, I don't seem to be able to figure this out. 
I will continue with relative includes for a while and see if I will come to some insights when I continue developing.

It is likely that it has something to do with how I am implementing my HAL, however I see no strange includes in that that could cause issues, it does what ArduinoHal does, basically. It should be noted, though that ArduinoHal is inside the library. 

There was a comment in /RadioLib/extras/test/SX126x/CMakeLists.txt about having to use add_subdirectory if one haven't built it as a shared library...but if I do that, it would seem that I'd break the dependency-driven model.

jgromes, 2023-05-17:
I don't see your HAL implementation in the repository you linked, so I can't really comment on that. In the meantime, I started writing my own ESP-IDF HAL and everything seems to be working so far - although it is without platformio. Once it's done I will add it to the NonArduino examples as a reference.

nicklasb, 2023-05-17:
OK. I have now recreated the problem in the above mentioned repo:
https://github.com/nicklasb/RadioLib_PIO_ESPIDF

The problem happens when the code #include:ing RadioLib.h is an ESP-IDF-component. 

I suppose that the solution has something to do with being able to refer to RadioLib as such, currently that doesn't seem possible.

nicklasb, 2023-05-17:
I tried adding an idf_component.yml, to make the component handle its own dependencies:
```
  # Required IDF version
  idf: ">=4.1"
  RadioLib:
    git: https://github.com/jgromes/RadioLib.git
```
But then the CMakeLists.txt for the library is configured as a project, so that causes an error. 
I will ask for guidance at the PlatformIO issues.

nicklasb, 2023-05-17:
So it would appear that in order to be used by ESP-IDF-components, RadioLib needs to have a CMakeLists.txt that works for that purpose. Not sure how that is done, though, obviously project(radiolib) can't be there, at least like that.

jgromes, 2023-05-21:
@nicklasb I added a functional ESP-IDF example - functional as in it builds, flashes and transmits via SX1278. Feel free to take a look at it, maybe you will be able to identify the issue you're seeing?

It's not using platformio, but I don't think that should be the issue - I didn't have to do any changes to the root RadioLib cmake to make it work (as per your previous comment).

nicklasb, 2023-05-21:
Hi,
The issue I was seeing WRT communication was solved, everything works now. 
Really well too, RadioLib is a really cool library, you've done great work!

WRT the component situation, the implementation has a relative path to the location of the library, so it doesn't help I am afraid as that is my current and very temporary workaround, and the CMakeList of RadioLib has a "project" definition. 

I proposed that one perhaps could use conditionals in the CMakeLists.txt, I know it is one-pass but I think that it should work.

The ESP-IDF hal is nice. Even though my HAL..eh...works...yours seems way more worked-through, I am not really an embedded dev by trade, so my bag of tricks isn't particularly full, there's a couple of gems there I'll reuse. :-)

jgromes, 2023-05-22:
> the CMakeList of RadioLib has a "project" definition.

I still don't understand why this is a problem. The project definition in RadioLib is not strictly necessary, so it can be removed, but I don't understand why is this causing trouble.

nicklasb, 2023-05-22:
> > the CMakeList of RadioLib has a "project" definition.
> 
> I still don't understand why this is a problem. The project definition in RadioLib is not strictly necessary, so it can be removed, but I don't understand why is this causing trouble.

Ok, I will try some variants. 
It causes trouble because I am getting an error if such configs are in components. (not scriptable and so on)

jgromes, 2023-05-23:
> It causes trouble because I am getting an error if such configs are in components. (not scriptable and so on)

Sorry, that's still not descriptive enough for me to understand. A working example would be nice, or at least the structury (e.g. - copy-paste RadioLib source to components/, make some change to the top-level CMake etc.). As far as I can tell right now there is no problem in the library itself, so I'm closing this issue for the moment. There clearly exists a way to build RadioLib for ESP-IDF.

Additionally, I don't think RadioLib CMake should include anything ESP-IDF specific, such as the idf_component command(s). Otherwise we would have to keep adapting the library to different build systems ...

closed 2023-05-23 by jgromes

jgromes, 2023-05-23:
> The problem is that an ESP-IDF component can not have a project() definition and other non-scriptable stuff in its CMakeLists.txt, but RadioLib does

I tried to approach a bit more to your configuration (RadioLib directly cloned into components/), removed the project() command from RadioLib root CMake, and the only thing that this changes is that not even add_library can be used. For me, having this is a hard requirement, because without it, CMake-based build system won't be able to use RadioLib as a ... well, library.

The ultimate problem is that ESP-IDF uses CMake in script mode and assumes components will be compatible with it. RadioLib obviously isn't, so the only thing that I can do at this level is to guard the CMake commands that cannot be used in script mode with `if(NOT(CMAKE_SCRIPT_MODE_FILE))` but that would effectively make the RadioLib root CMake empty, not sure if that helps.

nicklasb, 2023-05-23:
Yes, but in a parallel, and quite unrelated, discussion (https://github.com/ttlappalainen/NMEA2000_esp32/pull/12#issuecomment-1555244948) I saw this guy simply do 
`if(ESP_PLATFORM)` around the ESP stuff and the rest in the else().

I haven't tried that myself yet, won't have time for that until the weekend, but it seemed promising and worked for him.

nicklasb, 2023-06-04:
@jgromes,
Hi, I seems like that worked!
I forked this library, made a CMakeLists.txt with an `if(ESP_PLATFORM)` like above it works. 
Check out the https://github.com/nicklasb/RadioLib_PIO_ESPIDF repo, the master branch builds using this method.
Also, It seems like it keeps working doing a normal cmake build, so it doesn't break the library. 

Granted, I haven't tried with a huge number of projects yet, but at least this example library and my framework (that uses many more parts of RadioLib due to its HAL), works well like this. 
Running ESP-IDF, it adds a managed_components folder in the project. 

(Thanks @phatpaul for the inspiration!)

nicklasb, 2023-06-04:
@jgromes 
Oh, forgot to add the link to the fork:  https://github.com/nicklasb/RadioLib.git
The CMakeLists.txt is what is of interest, of course.

jgromes, 2023-06-04:
I'm not a huge fan of this solution, I don't think it should be the job of the library to adapt to non-standard build systems (and ESP-IDF does require calling idf_component_register). The sheer popularity of ESP platform somewhat forces my hand here, so I will implement it, however, with two modifications:

1. Instead of if/else, I will use if/return, same as is done in ArduinoJson ( https://github.com/bblanchon/ArduinoJson/blob/6.x/CMakeLists.txt). That way, if some other modification for some other non-standard build system is needed, it can be added without additional branching.

2. I will also add conditional error to be raised when the library is being compiled in script mode (after the ESP_PLATFORM condition, of course), since that is primary issue.

nicklasb, 2023-06-04:
@jgromes 
My view is that there are no standard build systems, they are just popular/unpopular. 
Basically because a build system is a way too complex a thing to standardize.
A standard then become a problem because it has to have its own developer community and handle all possible situations, ending up either too political, complicated or a hurdle that stops progress. 
Maybe I have been a bit burnt by "standards" in a lot of other languages and platforms, but that is how I see it. 

Sure, that is probably better, looks better, too.

jgromes, 2023-06-04:
@nicklasb added into the root CMake and also updated the example - thanks for investigating this!

nicklasb, 2023-06-04:
> @nicklasb added into the root CMake and also updated the example - thanks for investigating this!

Works great for me! Thanks for implementing!!