jgromes/RadioLib #752: Syncword and sx1276 and sx1262

nicklasb, 2023-05-19:
Hi!
I am having an issue sending between sx1276 to sx1262; it seems like I cannot get past the syncword check.
As I have the radio.getIrqStatus() function and both irqFlag and irqMask set  RADIOLIB_SX126X_IRQ_ALL on the sx1262, I can see I am getting radio.getIrqStatus() == 4, preamble detected on the sx1262, but I am not getting further. At some point I got 0x14, but still, that still doesn't include the syncword.

I have tried to make sense of the slightly confusing concept of sync word, with its reserved high nibbles or whatever, and it seems like having the sync word set to 0x12 on both sides *should* make it work. 
However, It doesn't work. And now I don't really know what to do.

nicklasb, 2023-05-19:
> You didn't provide much information, so I guessed that you're trying to transmit from SX1276 to SX1262. I tried it using just the default configuration (`begin()` on both sides) and I'm receiving every single packet. The SNR was > 11 dB, so to rule out the signal was strong enough to beat the sensitivity impact from bad sync word, I set the SX1262 sync word to 0x34. Sure enough, no packets at all. Setting it back to 0x12 restores the communication.
> 
> Sooo ... please follow the issue template ;) code, debug output, hardware setup, all that stuff.

Thanks for your reply!
Actually you answered the question I realized I just could've asked; setting the same sync word on both sides *should* make it work. 
But I could actually not find that statement anywhere, just a lot of things that could mean that or something else. :-)
I have previously had it up and working, but then with a different library. 

With defaults it doesn't work for med. As I have my own HAL, it could also be a whole bunch of other stuff, which means I will need to re-verify that stuff. I'll just know this now and continue huffing and puffing. :-)

closed 2023-05-19 by nicklasb

jgromes, 2023-05-19:
The issue on SX126x is that the sync word is padded with nibbles that "do other stuff" according to an official Semtech communication. RadioLib works around this by having the same `setSyncWord` interface on both SX126x and SX127x. I don't know what value are you using, but it should only be either 0x12 or 0x34. Other values may or may not work any may or may not have impact on sensitity - Semtech response is that those are the only two options, even if the datasheets do not even hint at that.

nicklasb, 2023-05-20:
It was something different of course, I did resets the states of things too often, basically it was a timing issue.
Thanks for your support!