jgromes/RadioLib #753: LLCC68，get error : 'RADIOLIB_ERR_CHIP_NOT_FOUND,with release6.0.0

bjfhskj, 2023-05-25:
With module LLCC68, when I use 6.0.0 or 5.7.0，I get error : 'RADIOLIB_ERR_CHIP_NOT_FOUND, but 5.6.0 all is right,
MCU : ESP32
 Arduino IDE version:2.1.0
Code:
#include <RadioLib.h>
SPIClass mySpi (HSPI);
SPISettings spiSettings (2000000, MSBFIRST, SPI_MODE0);
SX1262 radio = new Module (15, 16, 4, 17, mySpi, spiSettings);
mySpi.begin ();
  int state = radio.begin(868.0, 125.0, 9, 7, 0x12, 18, 8, 0, false);
  if (state == RADIOLIB_ERR_NONE) {
    //Serial.println(F("success!"));
  } else {
    //Serial.print(F("failed, code "));
   Serial.println(state);   ////get error -2, RADIOLIB_ERR_CHIP_NOT_FOUND
    while (true);
  }

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please use Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jgromes, 2023-05-25:
As stated in the issue template - please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

jgromes, 2023-06-15:
Closing due to inactivity

closed 2023-06-15 by jgromes