jgromes/RadioLib #754: CS port not dected and erro code -2

Anhilator, 2023-05-29:
I am using this library to communicate using a module E07-M1101D-SMA which is basically an rf transceiver that use the cc1101 chip, which use the following pin schema:
[image]

I am currently using this configuration:
// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  2
// RST pin:   unused
// GDO2 pin:  3
CC1101 radio = new Module(10, 2, RADIOLIB_NC, 3);

So I linked pin 1 and 2 with ground and vcc, pin 3 with GDO0 pin at pin 2 on the arduino, pin 4 with CS at pin 10 on the arduino and pin 8 on the GDO2 on pin 3 on arduino.
But the device is not being recognized during the initilization giving an error code -2 despite the right links.
Any ideas?

Anhilator, 2023-05-30:
Thank you for response, I am currently using Arduino UNO R3, Radiolib version 6.0.0, the code is the following (it's an example provided by the repo):

// include the library
#include <RadioLib.h>

// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  2
// RST pin:   unused
// GDO2 pin:  3
CC1101 radio = new Module(10, 2, RADIOLIB_NC, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//CC1101 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize CC1101 with default settings
  
  Serial.print(F("[CC1101] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  
}

void loop() {
  Serial.print(F("[CC1101] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to 63 characters long
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 63 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 64 bytes
    Serial.println(F("too long!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(100);
}

The full wiring is (on the left the RF transceiver and on the right the arduino):
GRN (PIN 1) -> GRN
VCC (PIN 2) -> 3.3 V
GDO0 (PIN3) -> Digital pin 2
CSN (PIN 4) -> Digital pin 10
SCK (PIN 5) -> Digital pin 13
GDO2 (PIN 8) -> Digital pin 3

The rest of the pin are not connected. I didn't modified the debug mode so it's on default.
Thanks in advance for your help.

jgromes, 2023-05-30:
> The rest of the pin are not connected

In that case, you did not connect most of the SPI interface - the MISO and MOSI pins must also be connected. See e.g. https://en.wikipedia.org/wiki/Serial_Peripheral_Interface

On Arduino Uno, MOSI is pin 11 and MISO is pin 12.

Anhilator, 2023-05-30:
Great! I was wondering another thing: is any way to modify the receiving timeout for the lib or it is defined by the hardware configuration of the antenna? I am asking this since while I was trying to receive a signal using the standard example for receiving, the communication timed out almost every second and I thought it could be disruptive for the communication having such a small timeout making difficult the communication since while a signal was arriving the timeout make it impossible to be received. Am I wrong in the reasoning or it's right?

jgromes, 2023-05-30:
I take it the issue was resolved ...?

Regarding timeout, it's controlled by the hardware, so I would suggest using interrupt-driven receive instead, which does not block execution, hence no timeout.

Anhilator, 2023-05-30:
perfect, thanks for the support, really appreciate it.
The issue is resolved, thank you so much for your help

closed 2023-05-30 by Anhilator