jgromes/RadioLib #759: CC1101 on WeMos D1 mini not detected

mpw96, 2023-05-30:
Hi,
I'm afraid I cannot get my ESP to successfully execute the radio.begin() method. Here's my setup:

```c++
#include <RadioLib.h>

CC1101 radio = new Module(15, 4, RADIOLIB_NC, 2);

void setup() {
  Serial.begin(115200);
  delay(1000);
  // initialize CC1101 with default settings
  bool connected = false;
  while(!connected) {
    Serial.print(F("[CC1101] Initializing ... "));
    int state = radio.begin();
    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
        connected = true;
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        delay(1000);
    }
  }
}

void loop() {
  Serial.print(F("[CC1101] Waiting for incoming transmission ... "));
}```

**Hardware setup**
![CC1101-868mhz-radio-module-pinout](https://github.com/jgromes/RadioLib/assets/12065222/0150cba5-b880-481c-accb-9513744ed446)

**Debug mode output**
```
[CC1101] Initializing ... CC1101 not found! (1 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (2 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (3 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (4 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (5 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (6 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (7 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (8 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (9 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
CC1101 not found! (10 of 10 tries) RADIOLIB_CC1101_REG_VERSION == 0x00FF, expected 0x0004/0x0014
No CC1101 found!
failed, code -2
```

**Additional info (please complete):**
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.2.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
 
RadioLib version 6.0.0

mpw96, 2023-05-30:
Oh, and thank you for this great library, @jgromes!

jgromes, 2023-05-31:
Looks like the SPI always returns 0xFF, which is typical for disconnected wires or incorrect pinout - could you check the pins are correct? WeMos seems to have different pins number on the board compared to the GPIOx pin number (e.g. GPIO15 is labeled D8 on the board).

mpw96, 2023-05-31:
> I will next check continuity of the wires and post the result.

The wires themselves seem okay, too. :-(

jgromes, 2023-05-31:
In that case, the next thing to check is whether the default SPI instance (what the WeMos D1 board in Arduino framework calls `SPI`) actually matches the hardware. ESP8266 can change on which pins the SPI is located, and you can provide a different SPI instance to RadioLib. Usually I would check this with an oscilloscope (if CLK really has clock, if the MOSI/MOSI really have data etc.), not sure if that is an option for you.

jgromes, 2023-06-15:
Closed due to inactivity, feel free to reopen later.

closed 2023-06-15 by jgromes