jgromes/RadioLib #762: measured period of startTransmit to tx_done interrupt is unrelated to payload size

yonatanirv, 2023-06-01:
hey,
I'm trying to evaluate the Tx time of a Lora TX message.
I've been using the "RadioLib SX128x Transmit with Interrupts Example" and added a timer to measure the time between the startTransmit and the TX done interrupt flag change (transmittedFlag ). 
what I can't figure out is that the recorded time is always ~152 ms no matter what is the payload size. I'm using the default packet parameter values.
the measured value DOES change when I set different SF values.
Maybe I'm missing something or not fully understanding how the IRQ/DIO interrupts are triggered, please let me know what I'm missing and if there is another way to get the airtime value for a TXed message.

Thanks!

`/*
   RadioLib SX128x Transmit with Interrupts Example

   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX128x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx128x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

#include <RadioLib.h>
#include "boards.h"

SX1280 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

// save transmission state between loops
int transmissionState = RADIOLIB_ERR_NONE;

// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

uint32_t counter = 0;

uint32_t timer;

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void)
{
    // check if the interrupt is enabled
    if (!enableInterrupt) {
        return;
    }

    // we sent a packet, set the flag
    transmittedFlag = true;
}

void setup()
{
    
    initBoard();
    // When the power is turned on, a delay is required.
    delay(1500);

    // initialize SX1280 with default settings
    Serial.begin(115200);
    Serial.print(F("[SX1280] Initializing ... "));
    int state = radio.begin();
#ifdef HAS_DISPLAY
    if (u8g2) {
        if (state != RADIOLIB_ERR_NONE) {
            u8g2->clearBuffer();
            u8g2->drawStr(0, 12, "Initializing: FAIL!");
            u8g2->sendBuffer();
        }
    }
#endif
    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true);
    }

    // set output power to 3 dBm    !!Cannot be greater than 3dbm!!
    if (radio.setOutputPower(3) == RADIOLIB_ERR_INVALID_OUTPUT_POWER) {
        Serial.println(F("Selected output power is invalid for this module!"));
        while (true);
    }

    // set spreading factor to 10
    if (radio.setSpreadingFactor(8) == RADIOLIB_ERR_INVALID_SPREADING_FACTOR) {
        Serial.println(F("Selected spreading factor is invalid for this module!"));
        while (true);
    }

    // set the function that will be called
    // when packet transmission is finished
    radio.setDio1Action(setFlag);

    // start transmitting the first packet
    Serial.print(F("[SX1280] Sending first packet ... "));

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    // transmissionState = radio.startTransmit("Hello World!");

    // you can also transmit byte array up to 256 bytes long
    // byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
    //                   0x89, 0xAB, 0xCD, 0xEF
    //                  };
    // state = radio.startTransmit(byteArr, 8);

    transmissionState = radio.startTransmit((uint8_t *)&counter, 4);
}

void loop()
{
    delay(1000);
    Serial.println("here");
// check if the previous transmission finished
    if (transmittedFlag) {
        // disable the interrupt service routine while
        // processing the data
        enableInterrupt = false;

        // reset flag
        transmittedFlag = false;

        if (transmissionState == RADIOLIB_ERR_NONE) {
            // packet was successfully sent
            Serial.println(F("transmission finished!"));

            // NOTE: when using interrupt-driven transmit method,
            //       it is not possible to automatically measure
            //       transmission data rate using getDataRate()
#ifdef HAS_DISPLAY
            if (u8g2) {
                u8g2->clearBuffer();
                u8g2->drawStr(0, 12, "Transmitting: OK!");
                u8g2->drawStr(0, 30, ("TX:" + String(counter)).c_str());
                u8g2->sendBuffer();
            }
#endif
        } else {
            Serial.print(F("failed, code "));
            Serial.println(transmissionState);
        }

        // wait a second before transmitting again
        delay(50);

        // send another one
        Serial.print(F("[SX1280] Sending another packet ... "));

        // you can transmit C-string or Arduino string up to
        // 256 characters long
        // transmissionState = radio.startTransmit("Hello World!");

        // you can also transmit byte array up to 256 bytes long
        /*
          byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                            0x89, 0xAB, 0xCD, 0xEF};
          int state = radio.startTransmit(byteArr, 8);
        */

        // transmissionState = radio.startTransmit((uint8_t *)&counter, 4);
        
        // ~200 bytes to be TXed 
        transmissionState = radio.startTransmit("12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890");

        counter++;
        // we're ready to send more packets,
        // enable interrupt service routine
        enableInterrupt = true;
        timer = millis();
        while(transmittedFlag == false)
        {
          delay(1);
        }
        Serial.print("TX done: ");
        Serial.println(millis()-timer);
    }
}
`

jgromes, 2023-06-01:
How much are you varying the payload size? If it's only by a couple bytes, then the difference could either be small enough to not be registered (the precision of millis() can be quite bad, depending on what board you have). Additonally, LoRa encodes bits as symbols (number of bits encoded varies with the SF), so it could happen that payloads of slightly different size will have the same number of symbols.

jgromes, 2023-06-01:
That's interesting, as far as I can tell the interrupt should be set to fire when transmission is finished. Is tehre a difference if you use the `micros()` function? Overall, I can't think of much that would influence this, though I would propose to change the measurement to the following:

```c++
enableInterrupt = true;
uint32_t start = millis();
while(transmittedFlag == false) {
  // only needed so that watchdog does not trigger for long packets
  yield();
}
uint32_t length = millis() - start;
Serial.println(length);
```

That way the measurement is not influenced by the 1 ms delay in the waiting loop and printing to Serial.

jgromes, 2023-06-04:
I tried to replicate this behavior with SX1280 and Arduino Pro Mini, and unfortunately I cannot do it. Transmitting 200 bytes takes about 214 ms according to the scope (it's a bit arguable where to place the starting point when monitoring the DIO pin directly), micros() value is 212.8 - 212.9 ms. The SX1280 calculator claims 213 ms.

Transmitting the default "Hello World!", scope says 28.8 ms, micros() timer 28.3 ms and SX1280 calculator results in 28.5 ms, so everything is consistent. So as far as I can tell, the interrupt fires correctly.

yonatanirv, 2023-06-04:
thanks for looking into that and testing it.
I can't really think why my outcome is different but I'll keep trying and update soon.

jgromes, 2023-06-15:
Closing this because I was not able to replicate it - feel free to reopen if new information that points to a problem in the library becomes available.