jgromes/RadioLib #764: loop() gets stuck in receive() method

timmornYE, 2023-06-02:
I have a strange behavior. I have sender and receiver. The sender works endless, but the receiver gets stuck if I move my hand over the LoRa Module.

## Reproduce

I used code from [RadioLib/examples/SX127x/SX127x_Receive/](https://github.com/jgromes/RadioLib/tree/master/examples/SX127x/SX127x_Receive) and changed the following:

Line 31 to:
```c
RFM95 radio = new Module(10, 3, 5, 6);
```

Because I have this module: https://docs.rs-online.com/ea5c/0900766b81544026.pdf

Line 42 I changed to:
```c
int state = radio.begin(868.0, 125.0, 12, 7, 0x34, 20, 8, 1);
```

For debugging purpose I changed Line 60 to:
```c
Serial.println(("Receive block"));
int state = radio.receive(str);
Serial.println(("Received"));
```

## Bug

All works well for hours. But if I move my hand over the LoRa module without touching it, Arduino stops loop without any error message to Serial. It always stops reproducible at "Receive block". This means the error happens somewhere within receive() method of this lib. 

That Arduino stops looping and not only my computer stops serial printing I can confirm because I also added a `Serial.println(("Still in Loop"));` after start of loop() method and Arduino also stops flashing the led while the sender keeps working all the time. 

## What I tried

I changed module of sender and receiver. But still sender works endless and receiver stops when I move my hand over the module.

## Question

Any suggestions for further debugging?

timmornYE, 2023-06-02:
There has to be some error in my code, because changed back to original and it worked.

timmornYE, 2023-06-02:
Ok, reopen. Didn't try long enough. Even with the original code definitely only changed commented code the same happens if I move my hand over the sensor. I think I produce bad bits doing so and then there happens some String overflow or similar because `\0` is missing or similar.

jgromes, 2023-06-04:
This is pretty strange. Normally I would recommend to run this in debug mode (https://github.com/jgromes/RadioLib/wiki/Debug-mode - feel free to try this), but I don't think you'll learn much that way.

> I think I produce bad bits doing so and then there happens some String overflow because \0 is missing or similar.

Seems highly unlikely. Bitflips are expected, and LoRa does a very good job of correcting them, that's partially why it has such good range. The packet length is normally sent with the packet and if some part of the decoding fails, some error would be returned.

To me, this seems mostly like a hardware issue, likely with supply power to the module/Arduino. Coincidentally, which Arduino are you using?

timmornYE, 2023-06-04:
Yeah, I also don't think so now, because like I wrote I also tried to send with length argument. Would I see it with debug mode if there is some overflow?

I use an Arduino UNO for the first tests because i think it is the most stable. I tried to lower output power to 12 because I also thought in this direction. But I was able to reproduce it also, although it seemed to need more "disturbing" signal. But what I didn't try is to flip the Arduinos. I only flipped the modules. Will try this. Perhaps it is somewhat damaged.

jgromes, 2023-06-04:
> Would I see it with debug mode if there is some overflow?

You will see the SPI transactions, so you will know all the commands and replies going to/from the transceiver. I'm nearly certain this issue is not due to any overflow in the library.

> I use an Arduino UNO for the first tests

Then you are using hardware with 5V logic (Arduino Uno) to interface with hardware with 3.3V logic (the RFM95 module), that is likely to damage it if you don't have a level shifter between the two.

> I tried to lower output power to 12

Then you have actually increased it by 2 dBm, since the default is 10.

timmornYE, 2023-06-04:
> Then you have actually increased it by 2 dBm, since the default is 10.

no, used 20 before (like written in introduction)

> Then you are using hardware with 5V logic

Oh, didn't think of this. For powering I used 3.3 V out, but you are right... logic is 5 V. I will use an 3.3V Arduino Pro Mini I have here for further testing.

closed 2023-06-08 by timmornYE