jgromes/RadioLib #771: Compile error: #include "<module_name>.h" under RadioLib

marbanak, 2023-06-14:
File: ModuleTemplate.cpp
Using Arduino IDE and SprakFun Hardware, nRF52840 Processor, controlling SX1276
Many thanks for you all you provided.

For compilation Using RadioLib, I have managed to resolve many compiler errors in #include calls, by refining the path.
But this following line eludes me. It is using a syntax, with which I am not familiar.
   #include "<module_name>.h"
I have tried relative paths of various kinds. I suspect this <module_name> is a string variable, which was not properly set somewhere.
Perhaps I am supposed to manually specify the module. If so, is tit the nRF controller, or the module under its control?
Anyone know what is going on, and how to fix it?
My apologies if this is something obvious.
Kind regards and gratitude.
    MABanak

jgromes, 2023-06-14:
First of all, ModuleTemplate.cpp is in the "extras" folder - why are you trying to compile it? As the name says, it's a template  to be used as a starting point when adding support for new wireless modules. It's not meant to be compiled.

Next: 

> I have managed to resolve many compiler errors in #include calls, by refining the path

What errors? It should be enough to just include RadioLib.h. If it doesn't work when you do that (wihtout any modifications to the library!), then your compiler is set up in a very strange way, which definitely shouldn't be the case if you are using Arduino IDE.

marbanak, 2023-06-15:
I humbly acknowledge your reply, jgromes.

My issues came down to managing a project under Arduino.
Renaming a project (sketch) requires the main .ino file to have the same name as the folder. Moving it around triggers a massive tar file download and the opening (and inclusion) of all files in RadioLib.
I saw something similar in an IDE from ATMEL years ago, and though the overhead looked strange, I figured it would take care of itself. I tried 3 times to make this all work.

In attempting to accurately describe all my steps, I replicated my experience a 4th time, this time staying in one folder with one project name. Well, there is no trouble now. With prior downloading of the RadioLib file, it looks like I'm OK now. Everything works.

Only 1 file is open in the IDE now, the .ino, and the compiler only pulls-in what it needs.

Once I cleared past the project management issues, a TypeDef.h file needed slight editing, whose contents had changed since the sample code was composed in 2021.
No problem!

This has been fixed manually in TypeDef.h, with explanatory comments. 
Other #defines must be watched for future use. 

**OLD RadioLib 4.6.0**
These are in the main code		
ERR_NONE 			= 0	
ERR_RX_TIMEOUT 	= -6	
ERR_CRC_MISMATCH 	= -7	

**RadioLib 5.0.0, 5.7.0 6.0.0**
These are the new defines
RADIOLIB_ERR_NONE 			(0)		
RADIOLIB_ERR_RX_TIMEOUT 		(-6)		
RADIOLIB_ERR_CRC_MISMATCH 	(-7)	

Here is an example change I made:
#define RADIOLIB_ERR_NONE                                      (0)
//MABanak added this to support legacy Sample Code 2023_06(Jun)_15
#define ERR_NONE                                      (0) 

Your contributions are impressive, sir.

Sincerest thanks,
Michael B

jgromes, 2023-06-16:
I suggest you stick to the way libraries are managed in Arduino IDE. In the Arduinoi IDE, go to Sketch -> Include Library -> Manage Libraries, search for RadioLib and then select the version you want to install - that's it, no need to mess with downloading/moving/renaming files.

If for some reason you have legace code, you can select a pre-5.0.0 version that will have the old defines, but I strongly suggest to update your sketch to use the new defines, so that you can get the latest udpates.

closed 2023-06-16 by jgromes