jgromes/RadioLib #773: radio.setDio0Action in v6.0.0?

K4KDR, 2023-06-20:
Hate to post this but after looking through the v6.0.0 source files & examples, I have to accept that I'm not going to know with any certainty (as a non-programmer) how my code needs to be adjusted.  And I really hate to just guess and perhaps 'think' my devices are still working when they perhaps are not.

After upgrading my Arduino library to v6.0.0, as the documentation clearly states, all my instances of "radio.setDio0Action" now generate an error.

Here is the example that is used in EVERY sketch that I use:

```
  #ifdef USE_SX126X
    radio.setDio1Action(onInterrupt);
  #else
    radio.setDio0Action(onInterrupt);
  #endif
```

... it's only a guess, but is the correct fix to replace each of these instances with:

```
  #ifdef USE_SX126X
    radio.setDio1Action(onInterrupt, RISING);
  #else
    radio.setDio0Action(onInterrupt, RISING);
  #endif
```

???

If not, could I trouble you to please direct me to the correct syntax?

If correct, how would I know whether it's appropriate to use RISING or (speculating...) 'FALLING'?

(... and for that matter, what does RISING & it's opposite even refer to?)

Very sorry to bother - if this is spelled out in the documentation or examples, I'm afraid that I wasn't able to find it.

Best regards,

-Scott,  K4KDR

jgromes, 2023-06-20:
Hi Scott, the release note is a bit unclear in this case, and it was pointed out by others before. The change was done only for the modules that expose the direction parameter, in this case SX127x. For the moduels that do not - SX126x - nothing changed. So for 6.0.0, it should be like this:

```c++
  #ifdef USE_SX126X
    radio.setDio1Action(onInterrupt);
  #else
    radio.setDio0Action(onInterrupt, RISING);
  #endif
```

One of the things I will do in near future for 6.1.0 will be to harmonize this interface for different modules, so that there no longer needs to be a `USE_SX126X` macro, and it can be done independently of the module type via the common `PhysicalLayer`, with additional example on how to use this abstraction. I will keep this issue open until then.

Regarding the direction, `RISING` should be used - it refers to the direction of the voltage level change that the interrupt is sensitive to. So when set to `RISING`, the action will be triggered when DIOx goes from logic low to high.

K4KDR, 2023-06-20:
Thank you SO much for taking the time to reply!

Well now I'm particularly glad that I asked since I was incorrect to change all lines including (unnecessarily) the one for `USE_SX126X` .

jgromes, 2023-06-21:
@K4KDR I added the generic methods, so now you can use `setPacketReceivedAction` and not have to worry about the radio type.

closed 2023-06-21 by jgromes

K4KDR, 2023-06-21:
Oh wow - thanks!

I'll check that out & learn how to use it!