jgromes/RadioLib #777: radio.receive(str) on 6.0.0 results in -705 but works on 5.7.0 for STM32 with Ebyte E22 SX1262

syund, 2023-06-24:
When using RadioLib 6.0.0, calling `int state = radio.receive(str);` results in a negative -705 error. When changing the library to version 5.7.0 and nothing else, receive works as expected. Receive also works as expected on 5.3.0. Transmit appears to work under both 6.0.0 an 5.7.0. I am using a nucleo l432kc board. I have also tried the latest commit from master [9472d5b](https://github.com/jgromes/RadioLib/commit/9472d5b63524bbba3f06ab418b67f53cc67753c4) and observed the same -705 behavior.

**Sketch that is causing the module fail**

```c++
#include <Arduino.h>
#include <RadioLib.h>

/*
   RadioLib SX126x Transmit Example
   This example transmits packets using SX1262 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)
   Other modules from SX126x family can also be used.
   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem
   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// Default SPI
// MOSI: D11
// MISO: D12
// SCK:  D13

// NSS pin:   D10
// DIO1 pin:  D5
// NRST pin:  D2
// BUSY pin:  D4
SX1262 radio = new Module(D10, D5, D2, D4);

void transmit_loop()
{
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE)
  {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));
  }
  else if (state == RADIOLIB_ERR_PACKET_TOO_LONG)
  {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));
  }
  else if (state == RADIOLIB_ERR_TX_TIMEOUT)
  {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));
  }
[... 485 lines cut ...]

failed, code -705
[SX1262] Waiting for incoming transmission ... CMD      80
DATW    0       A6

failed, code -705
[SX1262] Waiting for incoming transmission ... CMD      80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

CMD     11
DATR    0       A2      0       1

CMD     C0
DATR    0       22      0       22
```

**Additional info (please complete):**
 - MCU: Nucleo L432KC (STM32L432KC)
 - Link to Arduino core: https://github.com/stm32duino/Arduino_Core_STM32 
 - Wireless module type: SX1262 (Ebyte E22-900M22S]
 - Platformio IDE version: Core 6.1.7, Home 3.4.4
 - Library version: 6.0.0

jgromes, 2023-06-24:
I was able to replicate the issue using Arduino Uno + SX1268 with default configuration. From the debug output, it looks like the module returns SPI command timeout error after the setRx command. The interesting thing is that this is only happening with blocking receive method, so it seems like an issue either with the timeout, or with IRQ mapping.

Could you also provide debug for 5.7.0?

syund, 2023-06-24:
Sure thing!

<details>

<summary>5.7.0 Verbose debug output</summary>

```
--- Terminal on /dev/cu.usbmodem142203 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M       SX126x
CMD     80
DATW    0       AA

CMD     80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

CMD     80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

CMD     80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

CMD     17
DATR    0       A2      0       0       0       20

CMD     C0
DATR    0       22      0       22

CMD     7
DATW    0       A2      0       A2

CMD     C0
DATR    0       22      0       22

CMD     97
DATW    4       A2      0       A2      1       A2      40      A2

CMD     C0
DATR    0       22      0       22

CMD     8F
DATW    0       A2      0       A2

CMD     C0
DATR    0       22      0       22

CMD     8A
DATW    1       A2

CMD     C0
DATR    0       22      0       22

CMD     93
DATW    20      A2

CMD     C0
DATR    0       22      0       22
[... 258 lines cut ...]
DATR    0       22      0       22

CMD     D       7       36
DATW    D       A2

CMD     8C
DATW    0       A2      8       A2      0       A2      FF      A2      1       A2      0       A2

CMD     C0
DATR    0       22      0       22

CMD     82
DATW    0       A2      66      A2      66      A2

CMD     80
DATW    0       A6

CMD     12
DATR    0       A2      0       2       0       0

CMD     C0
DATR    0       22      0       22
```

</details>

jgromes, 2023-06-24:
Turns out this was a bit more complex than I thought, the bug was present even pre-6.0.0, but due to a combination of other factors it did not manifest. Anyway, it's fixed now (tested against a second Uno + SX1268), thanks for reporting!

closed 2023-06-24 by jgromes

syund, 2023-06-24:
Confirmed working on my setup as well! Thank you!