jgromes/RadioLib #783: Library should allow the use of all 3 Multi-Purpose Digital Input/Output for SX1280

SMH17, 2023-06-29:
**Describe the bug**
In [SX1280](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/2R000000HoCW/8EVYKPLcthcKCB_cKzApAc6Xf6tAHtn9.UKcOh7SNmg) there are 3 DIOs that can be configured as an interrupt. Any of the 3 DIOs can be selected as an external interrupt source for the transceiver and can be mapped to any interrupt signal from the transceiver.
This library supports only the use of DIO1, this limit implementation capabilities of SX1280.

**Expected behavior**
Library should support the optional use of other DIOs defined in SX1280.

**Screenshots**
[image]

**Additional info (please complete):**
 - Library version 6.1.0

jgromes, 2023-06-29:
> this limit implementation capabilities of SX1280

Can you provide a specific use-case for this? So far I have not seen that more than one DIO would be needed.

The SX126x also has 3 DIO pins, only DIO1 is supported as interrupt source. The SX127x has 6 DIO pins which all can cause an interrupt, but the library only supports interrupts for DIO0 and DIO1, simply because so far there was no use-case for the others.

SMH17, 2023-06-29:
@jgromes E.g., you can take advantage of them if you have to use more than 1 module for MIMO implementation; they are also useful if you have to implement a multisensor gateway or any other scenario where having only DIO1 is limiting. Certainly, not everyone needs all them, but they are there for very good reasons and library should support them as well.

jgromes, 2023-06-29:
Sorry, but I remain unconvinced - if you have a **specific** use-case that requires more than one DIO, then I'm fine with adding this support. But until then, it's a case of "wouldn't it be nice if ..." which typically leads to scope creep.

To me, specific use-case is "I need to have DIO1 set up to trigger X and simultanously DIO2 to trigger Y and it cannot be done by mapping both cases to just one pin because of XYZ". That would be a good argument for adding new features.

SMH17, 2023-06-29:
@jgromes It's not clear to me why you're so skeptical that having only 1 out 3 available DIOs is limiting, but I have noticed this limit because actually I need them for an implementation I'm working on. This library would have been very handy for prototyping tests I need, but the lack of support for the other DIOs does not allow me to use it, or at least not as is, unless I include support by myself. I would not have reported this otherwise. I cannot post the implementation details because it is a project for a thirdly party, covered by NDA.

jgromes, 2023-06-30:
The reason I'm asking is that if I know the background as to why you need some feature, then I can implement it in a way that fits well with the rest of the library and is also useful to others.

But I don't have this information, since you haven't provided it. I understand that you cannot post the actual source code, that's not what I want - I want to know the idea, the reason why you need this and what you are trying to do with it.

In the absence of that, the only thing I can do for you is to point you to the god mode macro:

https://github.com/jgromes/RadioLib/blob/a9a125a385972922f6c7f01e91c11e97e0fbcfa4/src/BuildOpt.h#L379-L386

If you enable it, you will have access to all methods, including the private `setDioIrqParams`.

https://github.com/jgromes/RadioLib/blob/a9a125a385972922f6c7f01e91c11e97e0fbcfa4/src/modules/SX128x/SX128x.cpp#L1435-L1441

This is a direct implemention of the SX1280 SetDioIrqParams command.

closed 2023-06-30 by jgromes