jgromes/RadioLib #784: LLCC68: Calibration failed, device errors: 0x20 failed, code -707

sigmashig, 2023-06-30:
**Sketch that is causing the module fail**

```c++
#include <Arduino.h>
#include <RadioLib.h>

LLCC68 radio = new Module(32, 33, 27, 25);

void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.println("[LORA LLCC68] Initializing ... ");
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE)
  {
    Serial.println(F("success!"));
  }
  else
  {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }
}

void loop() {
}

```

**Hardware setup**
I have tested several modules with chip LLCC68: E220-400M30S and RA-01SC. Both failed with the same error. Simultaneously, this script is working well with RA-02 (SX1278) with using correspondent class. So, I'm sure that I wire it correctly and SPI is working well.

ESP32-WROOM DevKit
Wiring:
MOSI - GPIO23 (VSPI MOSI)
MISO - GPIO19 (VSPI MISO)
SCK - GPIO18 (VSPI CLK)
NSS - GPIO32
DIO1 - GPIO33
BUSY - GPIO25
NRST - GPIO27 

**Debug mode output**
```
[LORA LLCC68] Initializing ...
CMDW    80
SI      0
SO      A2
CMDR    C0
SI      0       0
SO      22      22
CMDR    1D      3       20
SI      0       0       0       0       0    00       0       0       0       0       0    00       0       0       0
SO      A2      4C      4C      43      43   36       38      20      56      32      48   20       32      48      30      32      0     
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 4c 4c 43 43 36 38 20 56 32 48 20 32 48 30 32 00 | LLCC68 V2H 2H02.

M       SX126x    
CMDW    80        
SI      0
SO      A2        
CMDR    C0        
SI      0       0 
SO      22      22
CMDW    80        
SI      0
SO      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    80
SI      0
SO      A2
CMDR    C0
SI      0       0
SO      22      22
CMDR    17
SI      0       0       0
SO      A2      0       0
CMDR    C0
SI      0       0
SO      22      22
CMDW    97
SI      0       0       1       40
SO      A2      A2      A2      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    8F
SI      0       0
SO      A2      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    8A
SI      1
SO      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    93
SI      20
SO      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    88
SI      3       16      A       0       0    00
SO      A2      A2      A2      A2      A2   A2       A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    2
SI      43      FF
SO      A2      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    8
SI      0       0       0       0       0    00       0
SO      A2      A2      A2      A2      A2   A2       A2      A2
CMDR    C0
SI      0       0
SO      22      22
CMDW    89
SI      7F
SO      A2
CMDR    C0
SI      0       0
SO      2A      2A
CMDW    80
SI      0
SO      AA
CMDR    17
SI      0       0       0
SO      A2      0       20
CMDR    C0
SI      0       0
SO      22      22
Calibration failed, device errors: 0x20       
failed, code -707
```
**Additional info (please complete):**
 - MCU: ESP32-WROOM DevKit
 - Wireless module type: E220-400M30S (and RA-01SC)
 - Arduino IDE version:  1.8.13
 - Library version: 6.1.0 (the same problem exists for 6.0.0)

jgromes, 2023-06-30:
As per the LLCC68 datasheet, device error 0x20 is "XOSC failed to start". The default configuration for LLCC687 is to use TCXO with 1.6V reference voltage. Based on the error, I would guess that either your modules do not have a TCXO, or its reference voltage is different. Either way, it is configured by setting the `tcxoVoltage` argument of the `begin` method.

sigmashig, 2023-07-01:
Sure. The init string radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 0, false);  resolve the problem. As I can see, this problem is reproduced for a lot of LLCC68 modules. Maybe it makes sense to change the default to this parameter or add it to the Readme/header/... ?

closed 2023-07-01 by sigmashig

jgromes, 2023-07-01:
You are right that this pops up from time to time. Searching through the issues, I found a couple that were the same, so I changed the default configuration for LLCC68 to XTAL. We'll see how many people will have the opposite problem. BTW you can also do this by `radio.XTAL = true;` prior to `radio.begin()`. 

I also added a few comments to the SX126x/LLCC68 documentation - although I would like to point out that this is already exaplined in the documentation for the -707 error code: https://jgromes.github.io/RadioLib/group__status__codes.html#gabc695a4fae689e856ae6f618e334066f