jgromes/RadioLib #786: SX1278 + OOK RX + RadioLib > 5.7.0 issues

mag1024, 2023-07-02:
Hi,
I'm using a Lilygo/TTgo Lora32 V2.1/1.6 with a SX1278 module and the rtl_433_ESP library.
It works fine with RadioLib=5.7.0, but 6.0.0 and 6.1.0 don't seem to work (I was hoping to try to port to ESP-IDF so I'm looking to use the new HAL implementation).
The symptom I've observed is that while \w 5.7.0 the RegOpMode report 0x2d (RX Continuous?), with newer versions (and identical code "above" radiolib) the module initially reports 0x2c, and then goes to 0x29, or even 0x00. It still seems to receive the signals -- but the interrupt readout is garbage.

Obviously I'm not asking you to debug rtl_433_ESP, but I was hoping that, knowing what's changed since 5.7.0, you might have some pointers for how to approach debugging this.

FTR, this is the initialization sequence (as implemented [here](https://github.com/NorthernMan54/rtl_433_ESP/blob/master/src/rtl_433_ESP.cpp#L183)), in case anything seems odd:
```
rtl_433_ESP(6): SX1278 radio.begin() - success!
rtl_433_ESP(6): SX1278 setOOK - success!
rtl_433_ESP(6): SX1278 setCrcFiltering - success!
rtl_433_ESP(6): SX1278 setDataShapingOOK - success!
rtl_433_ESP(6): SX1278 OOK Thresh PEAK - success!
rtl_433_ESP(6): SX1278 OOK PEAK Thresh Decrement - success!
rtl_433_ESP(6): SX1278 Ook Peak Threshold Step - success!
rtl_433_ESP(6): SX1278 OokFixedThreshold - success!
rtl_433_ESP(6): SX1278 setBitRate - success!
rtl_433_ESP(6): SX1278 setRxBandwidth - success!
rtl_433_ESP(6): SX1278 RSSI Smoothing - success!
rtl_433_ESP(6): SX1278 preamble detect off - success!
rtl_433_ESP(6): SX1278 setDirectSyncWord - success!
rtl_433_ESP(6): SX1278 disableBitSync - success!
rtl_433_ESP(6): SX1278 receiveDirect - success!
```

Many thanks in advance!

jgromes, 2023-07-02:
As you can see in the release notes, there were quite a few changes between 5.7.0 and 6.0.0/6.1.0: https://github.com/jgromes/RadioLib/releases

0x2c in RegOpMode corresponds to frequency synthesis Rx and 0x29 to standby, so my guess would be that the modules tries to get into Rx, but fails to do so.

Is there any chance you could put together a minimum working example? I tried to perform all the configuration as it is done in rtl_433 using just Arduino Uno + SX1278, but the result is that the module ends up with RegOpMode correctly set to 0x2D (Rx continuous), as one would expect.

```c++
  int state = radio.beginFSK();
  state |= radio.setOOK(true);
  state |= radio.setDataShapingOOK(2);
  state |= radio.setOokThresholdType(RADIOLIB_SX127X_OOK_THRESH_PEAK);
  state |= radio.setOokPeakThresholdDecrement(RADIOLIB_SX127X_OOK_PEAK_THRESH_DEC_1_1_CHIP);
  state |= radio.setOokPeakThresholdStep(RADIOLIB_SX127X_OOK_PEAK_THRESH_STEP_0_5_DB);
  state |= radio.setOokFixedOrFloorThreshold(15);
  state |= radio.setBitRate(1.2);
  state |= radio.setRxBandwidth(250);
  state |= radio.setRSSIConfig(RADIOLIB_SX127X_RSSI_SMOOTHING_SAMPLES_2, RADIOLIB_SX127X_OOK_AVERAGE_OFFSET_0_DB);
  state = radio.setDirectSyncWord(0, 0);

  // reg op mode is 0x29 initially from previous standby calls, and then 0x2D
  radio.receiveDirect();
```

mag1024, 2023-07-02:
Thank you very much for the response!
Working from the minimal example, I think I figured it out.
[This](https://github.com/jgromes/RadioLib/commit/09ab5f8073f7d9b2ca20492b2aa78fa9c3aa35a1#diff-2f43a1ab977b88f0d25a8ea1816d16a29440707f54f80d644a9c2e71c4c66c88R260) was actually a subtly-breaking change. By inserting the new default parameter in front of the old, it changed the meaning of `getRSSI(true)` s.t. it no longer disabled "putting radio in receive mode for the RSSI measurement" -- thus kicking the module out of RX Continuous. Tricky!

closed 2023-07-02 by mag1024

jgromes, 2023-07-02:
Glad you were able to resolve it, that's indeed rather tricky. But unfortuantely tehre was a reason for this order - I needed to unify the interface for SX127x/SX126x getRSSI.