jgromes/RadioLib #787: Failure initialization of SX1276 connected to Raspberry Pi 4

simone-s, 2023-07-04:
Dear All,

we have the following configuration:

- Raspberry Pi 4 
- LoRa module SX1276

they are connected as follows:

```
LoRa SX 1276     |       RPi4
----------------------------- 
	3.3V		Pin 1
	GND		Pin 9
	MISO		Pin 21
	MOSI		Pin 19
	SCK		Pin 23
	NSS		Pin 22
	RESET		Pin 11
	DIO0		Pin7
```

with this configuration, we are assuming to use SPI channel 0 on the RPi4.

We have started with your example:
https://github.com/jgromes/RadioLib/tree/master/examples/NonArduino/Raspberry

We modified the main.cpp as follows to accommodate the hardware configuration above:

```
...

// create a new instance of the HAL class
// use SPI channel 1, because on Waveshare LoRaWAN Hat,
// the SX1261 CS is connected to CE1
PiHal* hal = new PiHal(0);

// now we can create the radio module
// pinout corresponds to the Waveshare LoRaWAN Hat
// NSS pin:   7
// DIO1 pin:  17
// NRST pin:  22
// BUSY pin:  not connected
SX1276 radio = new Module(hal, 24, 27, 11, RADIOLIB_NC);				

...
```

This code compiles directly on our target, i.e., RPi.
Unfortunately, this example program fails while running the initialization phase reporting the ERROR NUMBER -2.

May we ask for your support in using RadioLib with our hardware configuration?

We are also interested in using RadioLib to set ASK or FSK modulation on this SX1276 module. Is it possible?

Thank you in advance for your support.

simone-s, 2023-07-04:
Update: we are using the last version of RadioLIb available on this repository.

jgromes, 2023-07-04:
I'm a little bit confused about your pinout: in the first table, you seem to be referring to the physical pin number on the pin header. However, some of them seem wrong
* NSS: physical pin 22 is GPIO25, not 24 
* RST: physical pin 11 is GPIO17, not 11
* DIO0: physical pin 7 is GPIO4, not 27

So without knowing your hardware, I'm guessing the initialization should look like this:

```c++
SX1276 radio = new Module(hal, 25, 17, 4, RADIOLIB_NC);	
```

> we are assuming to use SPI channel 0 on the RPi4.

You have connected the NSS to a random GPIO pin, not one of the SPI0_CEx pins (GPIO7 or GPIO8), so I don't know what you mean by that. I don't know how the pigpio library behaves when using something else than CEx for the NSS pin.

> We are also interested in using RadioLib to set ASK or FSK modulation on this SX1276 module. Is it possible?

SX127x cannot do ASK, it can do LoRa, FSK or OOK, all of which are supported by RadioLib.

simone-s, 2023-07-04:
Dear,

thank you for your answer.
We changed the NSS connection between the LoRa SX1276 and the RPi4.
Now we have

```

LoRa SX 1276     |       RPi4
----------------------------- 
	3.3V		Pin 1
	GND		Pin 9
	MISO		Pin 21
	MOSI		Pin 19
	SCK		Pin 23
	NSS		Pin 24
	RESET		Pin 11
	DIO0		Pin 7
```

We are now using the following initialization:

```
SX1276 radio = new Module(hal, 8, 17, 4, RADIOLIB_NC);	
```
Indeed, assuming that inside the Module function we should use the GPIOXX value and not the PINXX one, we connected the NSS to the GPIO8 (pin 24).
As reference we follow this pinout for our RPi4: https://pinout.xyz/#

We recompiled the NonArduino example code and something changed.
Now the example program fails while running the initialization phase reporting the ERROR NUMBER -16.

Do you have any suggestion?

Thanks.

jgromes, 2023-07-04:
I would suggest to follow the issue template, at least by providing the debug + verbose output: https://github.com/jgromes/RadioLib/wiki/Debug-mode

-16 is SPI write failed, meaning the value in SPI register did not update even though it should have. Debug mode will show you exactly which register has failed.

simone-s, 2023-07-04:
We recompiled in debug-mode and we got the follows

```
[SX1261] Initializing ... R	42	12	
M	SX127x
R	1	8	
R	1	F	
W	1	9	
R	1	8	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	8	
R	1	B	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	
R	1	F	

address:	0x1
bits:		2 0
value:		0x1
current:	0xF
mask:		0x7
new:		0x9
read:		0xF
failed, code -16
```

simone-s, 2023-07-04:
Dear,

the first question about the Raspberry example. 
Since we do not have the Waveshare LoraWAN, do we have to drive the GPIO18 in PiHAL.h?
Because that pin should be the SPI1 CE0.

```
  void init() override {
      // first initialise pigpio library
      gpioInitialise();

      // now the SPI
      spiBegin();

      // Waveshare LoRaWAN Hat also needs pin 18 to be pulled high to enable the radio
      gpioSetMode(18, PI_OUTPUT);
      gpioWrite(18, PI_HIGH);
    }

    void term() override {
      // stop the SPI
      spiEnd();

      // pull the enable pin low
      gpioSetMode(18, PI_OUTPUT);
      gpioWrite(18, PI_LOW);

      // finally, stop the pigpio library
      gpioTerminate();
    }

```
Second, now we are initializing the LoRa module with this. 

``` 
SX1276 radio = new Module(hal, 8, 17, 4, RADIOLIB_NC);	

```
and we receive on the RPi4 terminal:

```
[SX1261] Initializing ... M	SX127x
Symbol length: 0.008000 ms
Symbol length: 4.096000 ms
success!
[SX1261] Transmitting packet ... failed, code -5
[SX1261] Transmitting packet ... failed, code -5

```

Is it correct? We use only one module, so we don't have any receivers.

Thanks.

jgromes, 2023-07-04:
> Since we do not have the Waveshare LoraWAN, do we have to drive the GPIO18 in PiHAL.h?

as the comment says, this is something that's specific to Waveshare LoRaWAN hat, so no need for you to do that.

Next, error code -5 is transmit timeout, which likely means you do not have DIO0 connected.

It seems like this is not an issue in the lbirary, so I'm converting it to a discussion instead.