jgromes/RadioLib #798: No tx on Seeed Studio E5 module (STM32WLE5xxxx)

ixmeins, 2023-07-16:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
Using any of the supplied Lora TX examples for the Nucleo WL55 board with the rfswitch_pins[] array configured for PA4,PA5,PA5 allows the application to build and run on the target with no error messages returned by the driver but no RF is output from the SX126x. (Confirmed with R&S spectrum analyser in case it appeared at an unexpected frequency).

**To Reproduce**
Insert this code block into example sketches, select E5 module from official STM32duino v2.6.0 board list. 

// Setup for E5 module
static const uint32_t rfswitch_pins[] = {PA4, PA5, PA5};
static const Module::RfSwitchMode_t rfswitch_table[] = {
  {STM32WLx::MODE_IDLE,  {LOW,  LOW, LOW}},
  {STM32WLx::MODE_RX,    {HIGH, LOW, LOW}},
  {STM32WLx::MODE_TX_LP, {LOW, HIGH, HIGH}}, // No LP mode just  HP
  {STM32WLx::MODE_TX_HP, {LOW, HIGH, HIGH}},
  END_OF_MODE_TABLE,
};

**Expected behavior**
RF output from SX126x module as per radiolib supplied example files.

**Additional info (please complete):**
 - MCU: STM32WLE5J
 - Link to Arduino core: v2.6.0
 - Wireless module type SX126x - Seeed E5 - https://wiki.seeedstudio.com/LoRa-E5_STM32WLE5JC_Module/#gpio-configuration-overview
 - Arduino IDE version  2.1.1
 - Library version 6.1.0

Hardware works with my RIOT OS code and LoraWAN examples within Arduino 2.1.1

ixmeins, 2023-07-16:
Stupidly easy fix. setOutputPower() values below 15 generate no output on this module. Values of 15-22 work as expected.

jgromes, 2023-07-16:
This is an interesting corner case, maybe the `setOutputPower` method should return an error when either LP or HP is not available but the user requests power output in that range. It would have to be defined by the RF switch table, so seeing as you did define LP switch mode even though there actually isn't one it wouldn't have worked anyway, but I think that can be included in the examples and docs.

Also, right now you will not be able to set power lower than 15 dBm, even though the HP amplifier can actually go as low as -9 dBm.

hallard, 2023-07-16:
just my 2 cents it’s may be because E5 has no LP path just HP
https://forum.seeedstudio.com/t/lora-e5-fw-bug-tx-eu868-85ma-too-high-consumption/260126/8

RAK 3172 had same issue until they released the LP version 
https://forum.rakwireless.com/t/rak3172-too-much-consumption-in-transmit-eu868/4781

scratched my head a lot on this one I made some PR on mbed to get things to work

jgromes, 2023-07-16:
@hallard yes, that was also my conclusion.

@ixmeins fixed in the commit mentioned above, thanks for reporting!

closed 2023-07-16 by jgromes