jgromes/RadioLib #800: Arduino UNO R4 WiFi ::  invalid conversion from 'uint32_t {aka long unsigned int}' to 'PinMode' [-fpermissive]

K4KDR, 2023-07-20:
Greetings!  Thank you for already including support for the new Uno-R4 boards!

Environment:  Ubuntu 18.04, Arduino IDE 1.8.13, RadioLib 6.1.0, Arduino Uno R4 Boards v1.0.2
Hardware:   Dragino SX1278 LoRa shield on Arduino Uno R4 WiFi board

Will gladly make changes in the sketch if needed, but wanted to post this in case it was indicative of something more general in RadioLib that you would prefer to adjust (if the impact is more global and not specific to my project).  Thank you so much!

Error on compile of the same modified FossaSat LoRa RX flowgraph that I've used on the Uno R3 for several years w/ the following  output:
```
--------------------------
Arduino: 1.8.13 (Linux), Board: "Arduino UNO R4 WiFi"

/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp: In member function 'virtual void ArduinoHal::pinMode(uint32_t, uint32_t)':
/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:25:56: error: invalid conversion from 'uint32_t {aka long unsigned int}' to 'PinMode' [-fpermissive]
   ::pinMode(pin, RADIOLIB_ARDUINOHAL_PIN_MODE_CAST mode);
                                                        ^
In file included from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Interrupts.h:8:0,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/ArduinoAPI.h:29,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/Arduino.h:4,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/BuildOpt.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/TypeDef.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.h:2,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:1:
/home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Common.h:96:6: note:   initializing argument 2 of 'void pinMode(pin_size_t, PinMode)'
 void pinMode(pin_size_t pinNumber, PinMode pinMode);
      ^~~~~~~
/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp: In member function 'virtual void ArduinoHal::digitalWrite(uint32_t, uint32_t)':
/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:32:64: error: invalid conversion from 'uint32_t {aka long unsigned int}' to 'PinStatus' [-fpermissive]
   ::digitalWrite(pin, RADIOLIB_ARDUINOHAL_PIN_STATUS_CAST value);
                                                                ^
In file included from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Interrupts.h:8:0,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/ArduinoAPI.h:29,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/Arduino.h:4,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/BuildOpt.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/TypeDef.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.h:2,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:1:
/home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Common.h:97:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)'
 void digitalWrite(pin_size_t pinNumber, PinStatus status);
      ^~~~~~~~~~~~
/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp: In member function 'virtual void ArduinoHal::attachInterrupt(uint32_t, void (*)(), uint32_t)':
/home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:46:93: error: invalid conversion from 'uint32_t {aka long unsigned int}' to 'PinStatus' [-fpermissive]
   ::attachInterrupt(interruptNum, interruptCb,  RADIOLIB_ARDUINOHAL_INTERRUPT_MODE_CAST mode);
                                                                                             ^
In file included from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Interrupts.h:8:0,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/ArduinoAPI.h:29,
                 from /home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/Arduino.h:4,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/BuildOpt.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/TypeDef.h:6,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.h:2,
                 from /home/k4kdr/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:1:
/home/k4kdr/.arduino15/packages/arduino/hardware/renesas_uno/1.0.2/cores/arduino/api/Common.h:113:6: note:   initializing argument 3 of 'void attachInterrupt(pin_size_t, voidFuncPtr, PinStatus)'
 void attachInterrupt(pin_size_t interruptNumber, voidFuncPtr callback, PinStatus mode);
      ^~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino UNO R4 WiFi.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
--------------------------------------------------------------------------
```

jgromes, 2023-07-20:
I was not able to replicate this using the latest master.

Reneases support was added after 6.1.0 was released, so if you have the 6.1.0 update, then it doesn't have the support yet. Either you have to pull the latest version from the repository, or wait until the next update.

closed 2023-07-20 by jgromes

K4KDR, 2023-07-20:
Thank you SO MUCH! 

I saw it listed under **Supported Arduino platforms:**, but it never occurred to me that this applied to the current build & not necessarily full releases that had been distributed.    Appreciate the guidance!!!