jgromes/RadioLib #803: Using custom SPI pins for CC1101 with Raspberry Pi Pico

internetrelative, 2023-07-25:
I'm confused on where to get started using custom SPI pins, apologize if I've missed something.
I've edited the example transmit code for the CC1101.
gdo0 --> 14
gdo2 --> 15
```
.
.
#define SCK_PIN 2
#define MISO_PIN 4
#define MOSI_PIN 3

CC1101 radio = new Module(5 , 14, RADIOLIB_NC, 15, SPI, SPISettings(2000000, MSBFIRST, SPI_MODE0));
.
.
SPI.begin();
int state = radio.begin();
.
.
```
I got this error:
`[CC1101] Initializing ... failed, code -2`

jgromes, 2023-07-25:
You can enable debug mode to get more information abtou what exactly is failing: https://github.com/jgromes/RadioLib/wiki/Debug-mode

Also, are you sure you are using the SPI correctly? You have set the default SPI, SPI settings and SPI begin, so how exactly are you "using custom SPI pins"?

internetrelative, 2023-07-25:
Thanks for the reply, I do not want to use the default SPI pins, I want to set it according to 
```
#define SCK_PIN 2
#define MISO_PIN 4
#define MOSI_PIN 3
```
how would I do that?

jgromes, 2023-07-25:
> how would I do that

That's up to your chosen platform. I don't know which Arduino core for Raspberry Pi Pico you are using - the official one (https://github.com/arduino/ArduinoCore-mbed) or the unofficial one (https://github.com/earlephilhower/arduino-pico)?

Either way, you will have to search the documentation for the Arduino core you are using to find this out.

jgromes, 2023-08-16:
Closing due to inactivity.

closed 2023-08-16 by jgromes