jgromes/RadioLib #805: [RF69] (aka RFM69) truncates last BIT in tx process. Fix included.

ramaza, 2023-07-30:
I'm testing Adafruit Feather M0 boards which are a combination of Atmel SAMD21 with HopeRF RFM69HCW radio modules. The goal is a communication between two such modules at maximum range and minimum bitrate.

With the help of an SDR with FSK demodulator I could verify that during TX the very last bit is truncated so that not every packet gets through without corruption. The problem became more obvious when I tried the same FSK settings with the RF69 library from https://github.com/lowpowerlab where the problem never happened.

After many hours of investigation the fix for this bug seems quite clear. According to the datasheet from HopeRF this lines in RF69.h
```
#define RADIOLIB_RF69_SEQUENCER_OFF                             0b00000000
#define RADIOLIB_RF69_SEQUENCER_ON                              0b10000000
```
should be
```
#define RADIOLIB_RF69_SEQUENCER_OFF                             0b10000000
#define RADIOLIB_RF69_SEQUENCER_ON                              0b00000000
```

I'm wondering why nobody so far took notice of this severe issue. My explanation is that with a significantly slower MCU than an 48MHz Cortex M0 it might not happen. Because you just need to add an artificial delay of a few hundred microseconds at the right place in the code and the problem doesn't appear.

jgromes, 2023-07-30:
Tahnks for figuring this out! Would you be willing to send a pull request to fix this?

ramaza, 2023-07-31:
I will send a pull request. There's one or two other things in the code for RF69 that caught my attention while debugging this problem. Like overwriting ALL interrupt flags to clear them although only a few interrupt flags support this with RF69. Should I create separate issue?

jgromes, 2023-07-31:
Yes please, if there's something else create new issue(s). It's possible that there are more copy/paste errors (as was the likely case for the sequencer configuration).

PR #807 (2023-08-05): Fixed wrong register definition 
Defines for RADIOLIB_RF69_SEQUENCER_OFF and RADIOLIB_RF69_SEQUENCER_ON where mixed up. Fixes #805
closed 2023-08-05 by jgromes