jgromes/RadioLib #812: e32-400m30s (sx1278+PA) inverted SetOutputPower result

whoim2, 2023-08-19:
Strange SetOuptupPower behavior for e32-400m30s module (sx1278 + unknown PA). As part of the experiment, I set the power:
SetOutputPower(2), (6) and (12). At 2 I get a burning module, at 6 moderately hot, at 12 almost cold.
When transmitting, I set the TX_EN pin of the module and remove it when I receive a DIO0 interrupt.
Why can observational results be inverted relative to expectations?

jgromes, 2023-08-19:
I have tested and measured the output from the SX127x using RF power meter and I'm quite confident that the output power matches the configuration.

The expectation that hot module = lots of power output is incorrect, it just means that most of the power is being reflected back into your module (or something is shorted) and the SX1278 over-current protection might be kicking in at higher power. 

Couple of things to investigate:

1. Check the connection to antenna, if the module is getting hot it would suggest most of the power is reflecting back into it instead of radiating from the antenna.
2. Does the SX127x use RFO or PA_BOOST to connect the RF output to the PA? RadioLib assumes PA_BOOST by default.
3. If you don't know what external PA is used, the module is useless to you since the PA will be frequency-dependent and will require some minimum/maximum input power level it can work with.

whoim2, 2023-08-19:
Thanks for such a detailed answer!
Indeed, the rssi readings behave as expected, it is more stable at higher output power, and at the same time lower heating. I could not find information about the PA used and will try to find out the maximum possible gain value experimentally.
The default mode is used, which means PA Boost. I could not find information, what is the difference between them, if you tell me, I will be grateful. All I do apart from the library is enable 1 on the txen rxen pins of the module.
I will also ask the manufacturer about PA, thank you!

jgromes, 2023-08-20:
> I could not find information, what is the difference between them

Let's take a look in the SX1278 datasheet. On the first page, you will find the block diagram of the device. On it , you will find both PA_BOOST as well as RFO (I marked them in red in the picture below).

[image]

As you can see, those are two completely different pins - you can check the pinout of the SX1278 further in the datasheet. So if you use the wrong one, it might be unconnected on your module, and if you transmit from it, almost all of the power will be reflected back into the SX1278 damaging it.

On a side note, this doesn't seem like an issue in the library so far, so I will convert this into a discussion.