jgromes/RadioLib #814: unable to initialize LoraWAN US915 band on SX1262

mischief, 2023-08-20:
**Describe the bug**
LoraWAN can't be configured on SX1262 with US915 band because of invalid power parameters.

A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
```
SX1262 radio =
    new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);
LoRaWANNode node(&radio, &US915);
...
node.beginOTAA();
```

we range check power level at https://github.com/jgromes/RadioLib/blob/3e8636b1f2002b3d9f129482f0d6ad49220129e0/src/modules/SX126x/SX1262.cpp#L85, but US915 band sets it out of range at https://github.com/jgromes/RadioLib/blob/3e8636b1f2002b3d9f129482f0d6ad49220129e0/src/protocols/LoRaWAN/LoRaWANBands.cpp#L86.

**Expected behavior**
initialize LoraWAN

**Screenshots**
N/A

**Additional info (please complete):**
 - MCU: LilyGO T-Deck/ESP32S3
 - Link to Arduino core: https://github.com/espressif/arduino-esp32
 - Wireless module type SX1262
 - Arduino IDE version  platformio core 6.1.6
 - Library version master @ 3e8636b1f2002b3d9f129482f0d6ad49220129e0

jgromes, 2023-08-20:
You're right, this is indeed incorrect. The problem is that no off-the-shelf SX126x/7x devices can actually get to the 30 dBm called for in the LoRaWAN specification without external PA. I think this should be fixed by making the module initialize to the maximum power it can support and which is allowed in the band.

However, even with that fix, you will run into a bigger problem in US915 band, which is that channels in this band are defined using CFlist mask, and not yet supported in RadioLib. LoRaWAN support is something I'm actively working on, and it's not ready yet. Also, it's going to be quite challenging for me to test, since the only gateway hardware I have is for the EU868 band.

jgromes, 2023-08-20:
@mischief I fixed the original problem, but mask-type CFList remains unimplemented. I will keep this issue open until that changes - though of course, a pull request would be much appreciated (since again - I don't really ahve a way to test in US915 band).

mischief, 2023-08-20:
> @mischief I fixed the original problem, but mask-type CFList remains unimplemented. I will keep this issue open until that changes - though of course, a pull request would be much appreciated (since again - I don't really have a way to test in US915 band).

hi,

thanks for your quick work.

unfortunately i am totally a :baby: when it comes to Lora, so i have zero idea about how this is supposed to work and i don't think i can contribute much... but i would be happy to test anything if i can.

jgromes, 2023-09-12:
@mischief so in the end I was able to convince my LoRaWAN gateway to work in the US915 band for testing purposes, and US915 band seems to be working now (in the latest commit). You might have to set `LoRaWANNode::startChannel` and `LoRaWANNode::numChannels` to select the frequency channel range - see the updated examples for details.

closed 2023-09-12 by jgromes