jgromes/RadioLib #816: Communication issue between SX1276 and SX1262

lucadjc, 2023-08-27:
Hello,
I've a Heltec v2 Lora device based on SX1276 and that is acting in my scenario as sender.
I've a new Heltec v3 Lora device based  on SX1262 device that is acting as receiver.

lora seems up & running but messages are not received, I don't know what wrongly i'm doing.
thanks in advance for your support

### **Sender**
#include <ArduinoJson.hpp>
#include <ArduinoJson.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <ESPDateTime.h>
#include <RadioLib.h>
#include <SPI.h>
#include <Hash.h>

// - Pin configs - ESP32 -
#define ss 18   
#define rst  14 
#define dio0 26 
#define dio1 35 
SX1276 radio = new Module(ss, dio1, rst, dio0);

#define DHTPIN 4     // what digital pin the DHT22 is conected to
#define DHTTYPE DHT22   // there are multiple kinds of DHT sensors

DHT dht(DHTPIN, DHTTYPE);

int timeSinceLastRead = 0;
int tempAvailable = 0;
char temperatureMessage[256];

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  600        /* Time ESP32 will go to sleep (in seconds) */
RTC_DATA_ATTR int bootCount = 0;

//Used for read battery data
#define VBAT_PIN 39
#define BATTV_MAX    4.14     // maximum voltage of battery
#define BATTV_MIN    1.3     // what we regard as an empty battery
#define BATTV_LOW    1.06     // voltage considered to be low battery
int battery_level = 0;
int pin_value = 0;

void setup() {
    initSerial();
    initDHT22();
    initLoRaSender();
    batteryStatus();
    manageTemperature();
    //startSleep();
    
}

void batteryStatus() {
    pin_value = analogRead(VBAT_PIN);
    //Serial.println((String)"Raw:" + rawValue);

    pin_value = analogRead(VBAT_PIN);
    float voltage_level = (pin_value / static_cast<float>(4096)) * 2 * 1.115 * 3.3; // calculate voltage level
   //battery_level = (voltage_level / BATTV_MAX * 100);
    //Serial.println((String)"Raw:" + pin_value + " Voltage:" + voltage_level + "V Percent: " + (battery_level)+"%");

    if (pin_value >= 2400) battery_level = 100;
    else if (pin_value >= 2100)    // 85-100% range
        battery_level = map(pin_value, 2100, 2400, 85, 100);
    else if (pin_value >= 1800)    // 10-85% range
        battery_level = map(pin_value, 1800, 2100, 10, 85);
    else if (pin_value >= 1700)    // 0-10% range
[... 240 lines cut ...]

    // print frequency error
    Serial.print(F("[SX1262] Frequency error:\t"));
    Serial.print(radio.getFrequencyError());
    Serial.println(F(" Hz"));

  } else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    //Serial.println(F("timeout!"));

  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}

**Receiver Output**
[SX1262] Initializing ... success!
[SX1262] Waiting for incoming transmission ...

jgromes, 2023-08-27:
Your SX1276 is transmitting FSK and your SX126x is receiving LoRa, that will not work. You have to choose one of those.

Also, the measured data rate is reported as about 50 Mbps - not a realistic number, which would seem to suggest the meassage you're trying to send could be empty.

lucadjc, 2023-08-27:
Ciao,
SX1276 changing to 

int state = radio.begin();
radio.setFrequency(868.0);

the result is
09:08:09:587 -> 
{"sensor":"temperature","sensor_model":"DHT22","location":"cantina","humidy":"65.40","temperature":"26.00","heat_index":"27.13","battery_level":"70","pin_value":"2041"}
09:08:10:161 -> timeout!

jgromes, 2023-08-27:
If your transmitter is timing out, it suggest a wiring issue. This line is rather suspicious:

`SX1276 radio = new Module(ss, dio1, rst, dio0);` DIO0 should come first.

Thie doesn't seem t obe an issue in the library itself, hence I'm converting this thread to a discussion instead.