jgromes/RadioLib #822: 'LoRaWANNode' does not name a type

greneit, 2023-09-06:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

I am using this example to register my device
examples/LoRaWAN/LoRaWAN_End_Device/LoRaWAN_End_Device.ino

However, when I compile, I get the error:
'LoRaWANNode' does not name a type on line 40, which has: LoRaWANNode node(&radio, &EU868);

I have included the library radiolib 6.1.0

Any ideas?

jgromes, 2023-09-06:
Could you also include information about your build environment, as asked for by the issue template (see below)?

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

greneit, 2023-09-06:
MCU: Arduino Pro Mini 
    Processor: atmega328p (3.3v 8mhz)
   Board currently has this firmware: https://github.com/Makerfabs/Lora-Soil-Moisture-Sensor/tree/master/V3
    Wireless module type: SX1276
    Arduino IDE version 1.8.19
    Library version 6.1.0

greneit, 2023-09-06:
I think I know what the issue is - I am using the version 6.1.0 of this library.

This version does not contain:
https://github.com/jgromes/RadioLib/tree/master/src/protocols/LoRaWAN
https://github.com/jgromes/RadioLib/tree/master/examples/LoRaWAN/LoRaWAN_End_Device

So these are not present in my Arduino Library for RadioLib 6.1.0

However, when I see it on github, I can see these protocol and examples. Has the LoraWAN been removed from 6.1.0, if so which version can I use to access LoraWAN end device connection features?

Thank you.

jgromes, 2023-09-06:
LoRaWAN is work in progress and was not yet released in any update, that's why you don't see it in the release 6.1.0. However, don't expect it to work on ATmega328P, it has extermely limited RAM and flash space and is very slow. It's currently the lowest-end platform supported by RadioLib and in the future, that support might get dropped altogether.

greneit, 2023-09-07:
Thank you for your input. Appreciate it.