jgromes/RadioLib #823: Support 4FSK on CC1101 module

MrTalon63, 2023-09-06:
**Is your feature request related to a problem? Please describe.**
Lately I was pondering at CC1101 documentation and read that indeed support hardware 4FSK (on top of other) modulations, due to the nature of how the modulation works it would be handy for projects that require a little bit more bandwidth, as from what I understand the module currently works in standard FSK mode with ability to switch to OOK modulation.

**Describe the solution you'd like**
Ability to switch mode modulation type to 4FSK (or even others like MSK or ASK) just like we can do right now with OOK. I believe there would need to be an additional check as it only allows for 300kBaud maximum throughput compared to 500kBuad when using FSK.

**Describe alternatives you've considered**
It seems like [Elechouse CC1101](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/b8c6af4c7c2214cd77a4e9b2e2cb37b24b393605/ELECHOUSE_CC1101_SRC_DRV.cpp#L421C1-L421C1) driver already supports changing modulations to all available, but also seems much more complex than RadioLib.

**Additional context**
None

MrTalon63, 2023-09-07:
I am aware of the 4-FSK "Horus" transmitter.
As for advantages, from what I know and read specifically from CC1101 datasheet, it provides a better spectral efficiency and a tiny uplift in receiving performance. Other than that, I don't really see any other compelling advantages when looking at it with fresh of eyes other than giving an option to use it.
I'm also not 100% sure how much work such implementation would require, but from looking at the datasheet it only requires to change the register byte to a different setting as deviation and such should be already handled by the current setting.

jgromes, 2023-09-08:
I think I'll keep this open for now with a low priority - if anyone comes up with a good use case/reason to implement this (that cannot be achieved by existing modes), then I will add this faster. Otherwise, it will be further down my todo list.

jgromes, 2024-10-28:
I started messing around with the M17 protocol which uses 4FSK, and while my ultimate goal is to have it running on all RadioLib modules, having "native" 4FSK support on CC1101 will come in handy, so I added this. It can be started by calling `CC1101::beginFSK4`, which has identical arguments as `CC1101::begin`. So far I did not run into any usage differences between 2FSK and 4FSK on CC1101.

closed 2024-10-28 by jgromes