jgromes/RadioLib #838: SX1281 LoRa not reading/outputing right data, but reading right packet length

cyboerg42, 2023-09-30:
After trying to get the SX1281 LoRa Modem working I ran into a weird issue... The packet is received, CRC checks out, SNR/RSSI is alright but it seems to decode the message wrong. I'm sending 64 bytes of random data, starting with FF FF FF for testing/debugging - as meshtastic did weird things - but my receiver is reading the first (4!) bytes as 43 43 43 43. 

Any chances that someone here knows what's wrong with that?

[image]

My code is heavily influenced by your SX1280 examples, transmission looking fine - at least the byte array that gets passed to RadioLib.

<details>
<summary>RX test-code</summary> 

 ```
#include <RadioLib.h>

SPIClass newSPI(HSPI);
SX1281 radio = new Module(27, 37, 26, 36, newSPI);

void setup() {
  Serial.begin(9600);
  newSPI.setFrequency(500000);
  newSPI.begin(25, 33, 32, 27);
  sleep(1);
  
  Serial.print(F("[SX1280] Initializing ... "));
  int state = radio.begin();

  radio.setFrequency(2430.0);
  radio.setBandwidth(203.125);
  radio.setSpreadingFactor(12);
  radio.setCodingRate(7);
  radio.setOutputPower(0);
  radio.setHighSensitivityMode(true);
  radio.setRfSwitchPins(10, 14);

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  radio.setDio1Action(setFlag);

  Serial.print(F("[SX1280] Starting to listen ... "));
  state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

volatile bool receivedFlag = false;

#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // we got a packet, set the flag
  receivedFlag = true;
}

void printArray(byte byteArr[], int len) {
  for (int i = 0; i < len; i++) {
    Serial.print(byteArr[i], HEX); // Print each byte in hexadecimal
    Serial.print(" ");
  }
}

int i;

void loop() {
  if(receivedFlag) {
    receivedFlag = false;
    int len = radio.getPacketLength();
    //int len = 128;
    byte byteArr[len] = {};
    int state = radio.readData(byteArr, len);

    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("[SX1280] Received packet!"));
      Serial.print(F("[SX1280] Length:\t\t"));
      Serial.println(len);
      
      Serial.print(F("[SX1280] Data:\t\t"));
      printArray(byteArr, len);
      Serial.println("");
      
      Serial.print(F("[SX1280] RSSI:\t\t"));
      Serial.print(radio.getRSSI());
      Serial.println(F(" dBm"));
      Serial.print(F("[SX1280] SNR:\t\t"));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));
      Serial.print(F("[SX1280] Frequency Error:\t"));
      Serial.print(radio.getFrequencyError());
      Serial.println(F(" Hz"));
    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      Serial.println(F("CRC error!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
    }
    radio.startReceive();
  }
}
```
</details>

<details>
<summary>TX test-code </summary> 

```
#include <RadioLib.h>

SPIClass newSPI(HSPI);
SX1281 radio = new Module(27, 37, 26, 36, newSPI);

int transmissionState = RADIOLIB_ERR_NONE;
byte randomArray[64];

void generateRandomArray() {
  for (int i = 0; i < 64; i++) {
    randomArray[i] = random(256); // Generate a random byte
  }
}

void printArray() {
  Serial.println("Random 64-byte array:");
  for (int i = 0; i < 64; i++) {
    Serial.print(randomArray[i], HEX); // Print each byte in hexadecimal
    Serial.print(" ");
  }
  
  randomArray[0] = 255;
  randomArray[1] = 255;
  randomArray[2] = 255;
  
  Serial.println();
}

void setup() {
  Serial.begin(9600);
  newSPI.setFrequency(500000);
  newSPI.begin(25, 33, 32, 27);
  sleep(1);
  
  Serial.print(F("[SX1280] Initializing ... "));
  int state = radio.begin();

  radio.setFrequency(2430.0);
  radio.setBandwidth(203.125);
  radio.setSpreadingFactor(12);
  radio.setCodingRate(7);
  radio.setOutputPower(3);
  radio.setHighSensitivityMode(true);
  radio.setRfSwitchPins(10, 14);

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  radio.setPacketSentAction(setFlag);

  Serial.print(F("[SX1280] Sending first packet ... "));
  generateRandomArray();
  printArray();
  transmissionState = radio.startTransmit(randomArray, 64);
}

volatile bool transmittedFlag = false;

#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  transmittedFlag = true;
}

int count = 0;

void loop() {
  if(transmittedFlag) {
    transmittedFlag = false;

    if (transmissionState == RADIOLIB_ERR_NONE) {
      Serial.println(F("transmission finished!"));

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }
    
    radio.finishTransmit();

    delay(1000);

    Serial.print(F("[SX1280] Sending another packet ... "));
    generateRandomArray();
    printArray();
    transmissionState = radio.startTransmit(randomArray, 64);
  }
}
```
</details>

jgromes, 2023-09-30:
Interesting, couple of things to check out:

1. Is all configuration successful, or does some of it fail? You're not checking the return codes. Unlikely it would cause this behavior, but just to be sure.
2. Did you try running both devices in debug verbose mode? It will show all SPI transacations taking place, so this could help pinpoint if this is happening on the side of the transmitter or the receiver.

cyboerg42, 2023-09-30:
Thanks for your quick reply, it seems like the right message is sent via SPI on the transmitter side. RX is also receiving the hex string as printed by my function.. Configurations Checks are OK too.

<details>
<summary>TX Debug Output</summary>
```
[SX1280] Initializing ... M	SX128x
CMDW	80	
SI	0	
SO	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	80	
SI	0	
SO	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	8F	
SI	0	0	
SO	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	8A	
SI	1	
SO	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	88	
SI	60	
SO	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	96	
SI	1	
SO	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	86	
SI	B8	9D	89	
SO	43	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDR	3	
SI	0	0	
SO	43	1	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	8B	
SI	90	18	3	
SO	43	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDR	3	
SI	0	0	
SO	43	1	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	8B	
SI	90	18	3	
SO	43	43	43	
CMDR	C0	
[... 581 lines cut ...]
SI	40	62	0	2	0	0	0	0	
SO	43	43	43	43	43	43	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	8F	
SI	0	0	
SO	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	97	
SI	FF	FF	
SO	43	43	
CMDR	C0	
SI	0	0	
SO	40	43	
CMDW	82	
SI	0	FF	FF	
SO	43	43	43	
CMDR	C0	
SI	0	0	
SO	0	3
```
</details>

jgromes, 2023-09-30:
It looks like the SX128x is returning its status instead of the data bytes, which could point to an issue in the low-level SPI handling. It's a bit strange because the interface should be the same as for the SX126x, which I know is working. I will try to get some hardware up and running to test this for myself.

cyboerg42, 2023-09-30:
Thanks a lot, just ask if you need something tested on my hardware & I'll try a few things myself... But I'll stay tuned about your updates too. :+1:

Got the board definition from ExpressLRS if that helps... This was also the "original" firmware running on this board. For the Arduino IDE I'm using the UM TinyPico board target.

cyboerg42, 2023-10-01:
Update from my side, I tried to implement the getBufferOffset Function - as another driver got that before reading - just to see that it always stays at 0.. Now I've took a step back and generated a non-random 255-byte array. The Receiver only sees 67 bytes.. And I see some pattern there. With a fixed packet size (255) the buffer does also only contain 67-bytes.

<details>
<summary>TX</summary>

```
[SX1280] Sending another packet ... 255-byte array:
1 2 3 4 5 6 7 8 9 A B C D E F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF B0 B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF C0 C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF E0 E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF F0 F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FB FC FD FE FF 
transmission finished!
```

</details>

<details>
<summary>RX (detecting packet size)</summary>

```
[SX1280] Received packet!
[SX1280] Length:		67
[SX1280] Data:		1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 40 41 42 43 
[SX1280] RSSI:		-33.00 dBm
[SX1280] SNR:		0.50 dB
[SX1280] Frequency Error:	0.39 Hz
```

</details>

<details>
<summary>RX (fixed packet size)</summary>

```
[SX1280] Received packet!
[SX1280] Length:		255
[SX1280] Data:		1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 0 1 2 3 40 41 42 43 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
[SX1280] RSSI:		-33.00 dBm
[SX1280] SNR:		0.25 dB
[SX1280] Frequency Error:	151.52 Hz
```

</details>

cyboerg42, 2023-10-02:
[image]

Received another board in the mail today (ESP8266/SX1280 based) - so I can confirm that the TX on the SuperD ESP32 board works. The ESP8266 decodes it and shows valid data.

cyboerg42, 2023-10-02:
[image]

Got it working, the second radio on the SPI bus didn't play nicely. Sorry for bothering, this shall be a notice to other peoples using this board. Please move to discussions and I'll explain in more details there what went wrong.

closed 2023-10-02 by cyboerg42