jgromes/RadioLib #854: Can't use in esphome

vekexasia, 2023-10-21:
Hello starting from 5.7.0 i can no longer use RadioLib with esphome. There might be something due to the fact that a library.json was added in 5.7.0 release

https://github.com/jgromes/RadioLib/compare/5.6.0...5.7.0#diff-64cc7cdde6789dc1c1c4a6f406cd1ff6a6a00027097788e2d52fc27b0d75502e

The error that i'm getting is the following (when using 5.7.0+) 
```
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- SPI @ 2.0.0
|-- RadioLib @ 6.2.0
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
.....
In file included from .piolibdeps/weather-pm25-v2/RadioLib/src/ArduinoHal.cpp:1:
.piolibdeps/weather-pm25-v2/RadioLib/src/ArduinoHal.h:16:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
```

apparently SPI can not be found but it's getting installed by depgraph.

jgromes, 2023-10-23:
I'm not able to reproduce this issue when only building RadioLib with platformio. Is the issue only present when building as part of esphome? If so, please try to build just a plain RadioLib example. If the issue goes away, it would suggest some problem with esphome, rather than RadioLib.

Also, I will need to see at least your platformio.ini file.

jgromes, 2023-11-11:
Closing due to inactivity.

closed 2023-11-11 by jgromes

PaulSchulz, 2024-11-13:
In esphome I can successfully compile with RadioLib library by downgrading to 5.6.0, and 5.7.0 breaks (as mentioned).
eg. 

esphome:
  name: "lora-receiver"
  libraries:
    - "SPI"
    - "Ticker"
    - "SX126x-Arduino"
    #- RadioLib=https://github.com/jgromes/RadioLib#7.1.0
    - RadioLib=https://github.com/jgromes/RadioLib#5.6.0
    #- RadioLib=https://github.com/jgromes/RadioLib#5.7.0

(I am attempting to migrate some code that is currently using the "SX126x-Arduino" library to RadioLib.)