jgromes/RadioLib #858: Problems with the LoRaWAN example

lewisxhe, 2023-10-28:
Hi JanGromes, Thanks for making LoRaWAN work, cool.

I have a few questions.

1. Write in the example `LoRaWAN_End_Device`
```
     // prior to LoRaWAN 1.1.0, only a single "nwkKey" is used
     // when connecting to LoRaWAN 1.0 network, "appKey" will be disregarded
     // and can be set to NULL
```
In TTN if `LoRaWAN Version` is selected below `1.1.0`, then the valid one will be
`Appkey` is not `NetKey` as mentioned in the example

In actual use, if `appKey` is filled in with the AppKey generated in TTN, then a **MIC Mismatch** error will occur when joining the LoRo gateway.

If I fill in `nwkKey` as the AppKey generated by TTN, `appKey = NULL`, then I can see in the TTN terminal that it is allowed to be joined, but `state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);`
The error code returned is **-6** or **-1112**

I try to select `1.1.0` for `LoRaWAN Version` in TTN, and fill in the ` joinEUI, devEUI, nwkKey, appKey` generated by TTN accordingly. It can be added to TTN normally for the first time, but if the device is restarted and the TTN is used for the second time, The terminal will prompt `DevNonce`, the solution is to lower the LoRaWAN version, which can be found [here](
https://support.milesight-iot.com/support/solutions/articles/73000585630-lorawan-sensors-faq-july-2022#:~:text=Q1%3A%20When%20the%20sensor%20reconnects%20to%20the %20TTN,version%201.0.3%20on%20Milesight%20Sensor%20and%20TTN%20settings.)

jgromes, 2023-10-28:
Couple of thoughts:

1. Here's the key derivation scheme for LoRaWAN < 1.1.0, directly from the specification. Notice that "The device's AppKey is not used". This is incorrect in TTN's web interface, and is supported by your own result - if you set RadioLib NwkKey as the "AppKey" in TTN, then you can actually join.
[image]

2. Error code -1112 is `RADIOLIB_ERR_JOIN_NONCE_INVALID`. Not surprising, considering that the previous reported nonce is -1, aka 0xFFFFFFFF - that leads me to believe that you did not call `node.wipe()` when first running the example.

3. >  if the device is restarted and the TTN is used for the second time, The terminal will prompt DevNonce, the solution is to lower the LoRaWAN version

That's probably again caused by the unitialzied persistent storage.

lewisxhe, 2023-10-29:
JanGromes,
I tried adding node.wipe() for the first time to clear the information saved in EEPROM, but I still couldn't join.

Explanation, I use SX1276 and it runs well. No matter how many times I restart it, it can successfully join TTN. But when I use SX1262, I can only see it being added in TTN, but RadioLib returns -6.

lewisxhe, 2023-11-03:
The issue will be closed first. I am just providing you with feedback on the report I used. Thank you again for sharing. Currently, due to other projects, I am unable to continue testing. I will continue to investigate in my free time,

closed 2023-11-03 by lewisxhe