jgromes/RadioLib #864: Problems with LoRaWAN / RAK11300

nmaas87, 2023-11-02:
Hello there,
I am trying to get the RAK11300 integrated module working using the "normal" Arduino Pico Framework ( https://github.com/earlephilhower/arduino-pico/ ) and the latest dev/master commit from this library.

The RAK11300 is a module directly wiring an RP2040 core to SX1262 module ( https://docs.rakwireless.com/Product-Categories/WisDuo/RAK11300-Module/Datasheet/#description ). The description is sadly very vague, but I got more information to work with (especially the link between the RP2040 core and the SX1262): https://forum.rakwireless.com/t/rak11300-pinout-rp2040-to-sx1262/8414

Most important infos are here:
````
SPI_SCLK = 10;		   // LORA SPI CLK
SPI_MOSI = 11;		   // LORA SPI MOSI
SPI_MISO = 12;		   // LORA SPI MISO
SPI_NSS = 13;		   // LORA SPI CS
RESET = 14;		   // LORA RESET
BUSY = 15;		   // LORA SPI BUSY
TXEN = -1;		   // LORA ANTENNA TX ENABLE (e.g. eByte E22 module)
RXEN = 25;		   // LORA ANTENNA RX ENABLE (e.g. eByte E22 module)
DIO1 = 29;

The antenna switch direction is controlled by the SX1262 itself through DIO2.
The antenna switch power is controlled with GPIO25
It uses an TCXO
````

With that, I used the default example as shown.
Note that I had to re-configure the SPI1 interface, as its usually on different GPIOs on a stock RP2040.
With this, the module intializes successfully, however, trying to OTAA on TTN either with
a) 1.0.2, Reg Rev B (with AppKey set to false)
b) 1.1.0, Reg Rev B (with AppKey)
both does generate successful join-accept messages and forwards join-accept messages - however, the RAK11300 always ends up with a failed oota code -6

I am not sure yet if this is an error as shown in https://github.com/jgromes/RadioLib/issues/858 - or if its something about the RXEN that needs to be setup. I can hardwire the RXEN to HIGH or LOW - it does not change anything in the result.

As additional infos, here is the official repo which builds just for Arduino Mbed:
https://github.com/RAKWireless/RAK-RP-Arduino/tree/main
And there is also an AT firmware that can be built
https://github.com/RAKWireless/RAK11300-AT-Command-Firmware/tree/main/RAK11300-AT-Arduino

The AT firmware works without issues on the RAK11300 and successfully joins.

Maybe you have an idea what I am doing wrong?

Thanks!

Debug:
````
[SX1262] Initializing ... 
RadioLib Debug Info
Version:  6.2.0.0
Platform: Raspberry Pi Pico (unofficial)
Compiled: Nov  2 2023 19:40:26

CMDW	80	
SI	0	
SO	AA	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	3	20	
SI	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
SO	A2	53	58	31	32	36	31	20	56	32	44	20	32	44	30	32	0	
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M	SX126x
CMDW	80	
SI	0	
SO	AA	
CMDW	80	
[... 1113 lines cut ...]
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print SNR (Signal-to-Noise Ratio)
    Serial.print(F("[LoRaWAN] SNR:\t\t"));
    Serial.print(radio.getSNR());
    Serial.println(F(" dB"));

    // print frequency error
    Serial.print(F("[LoRaWAN] Frequency error:\t"));
    Serial.print(radio.getFrequencyError());
    Serial.println(F(" Hz"));
  
  } else if(state == RADIOLIB_ERR_RX_TIMEOUT) {
    Serial.println(F("timeout!"));
  
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  // wait before sending another packet
  delay(10000);
}
````

nmaas87, 2023-11-03:
Ok, I got TTN join and Chirpstack join both working now with the #865 PR.
TTN actually works without it, but Chirpstack will deny all joins/communication if wrong.

What does not work is as described by #858 - a re-join is not possible at all, if the connection is lost or we try to re-join from a fresh startup with .restoreOTAA(); I get a failed -1101 / DevNonce issues.

Following is the code part and log.

A second issue is the Antenna Switch which is on GPIO25, I finally figured out that it actually powers the antenna switch (no matter TX or RX) as described here: https://www.nico-maas.de/?p=2607 - Update to the Antenna Switch / GPIO25. Maybe you got a better idea on how to include this without having the global
````
pinMode(25, OUTPUT);
digitalWrite(25, HIGH); 
````
which does have the Antenna Switch powered at all times.
To be precise: DIO2 controls the "direction" (e.g. TX or RX), GPIO25 just powers the switch.
However, the switch should only get enabled if we actually want to RX or TX to keep the power profile as low as possible. Do you have any suggestions for that?

````
  /*
  // this can take up to 20 seconds, and requires a LoRaWAN gateway in range
  Serial.print(F("[LoRaWAN] Attempting over-the-air activation ... "));
  //state = node.beginOTAA(appEUI, devEUI, (uint8_t*)nwkKey, (uint8_t*)appKey);
  state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  //state = node.beginOTAA(joinEUI, devEUI, nwkKey, NULL);
  //state = node.beginAPB(devAddr, (uint8_t*)nwkSKey, (uint8_t*)appSKey);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success oota!"));
    //while(true);
  } else {
    Serial.print(F("failed oota, code "));
    Serial.println(state);
    while(true);
  }
  */
  
  // after the device has been activated,
  // network can be rejoined after device power cycle
  // by calling "begin"
  
    Serial.print(F("[LoRaWAN] Resuming previous session ... "));
    state = node.restoreOTAA();
    //state = node.begin();
    if(state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while(true);
    }
    
}
````

````
[SX1262] Initializing ... 
RadioLib Debug Info
Version:  6.2.0.0
Platform: Raspberry Pi Pico (unofficial)
Compiled: Nov  3 2023 19:45:29

CMDW	80	
SI	0	
SO	AA	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	3	20	
SI	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
SO	A2	53	58	31	32	36	31	20	56	32	44	20	32	44	30	32	0	
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M	SX126x
CMDW	80	
SI	0	
SO	AA	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	17	
SI	0	0	0	
SO	A2	0	20	
CMDR	C0	
SI	0	
SO	22	
CMDW	7	
SI	0	0	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	97	
SI	0	0	1	40	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	8F	
SI	0	0	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	8A	
SI	1	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	93	
SI	20	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	88	
SI	3	16	A	0	0	0	0	
SO	A2	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	2	
SI	43	FF	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	8	
SI	0	0	0	0	0	0	0	0	
SO	A2	A2	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	89	
SI	7F	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDW	8B	
SI	9	6	3	0	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	40	
SI	14	24	
SO	A2	A2	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	7	36	
SI	0	0	
SO	A2	D	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	36	
SI	D	
SO	A2	
CMDW	8C	
SI	0	8	0	FF	1	0	
SO	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	96	
SI	1	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	8	E7	
SI	18	
SO	A2	
CMDW	9D	
SI	1	
SO	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	7	36	
SI	0	0	
SO	A2	D	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	36	
SI	D	
SO	A2	
CMDW	8C	
SI	0	8	0	FF	1	0	
SO	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	7	36	
SI	0	0	
SO	A2	D	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	36	
SI	D	
SO	A2	
CMDW	8C	
SI	0	8	0	FF	1	0	
SO	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDW	8B	
SI	9	6	3	0	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDW	8B	
SI	9	4	3	0	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	98	
SI	6B	6F	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	86	
SI	1B	20	0	0	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	8	D8	
SI	0	0	
SO	A2	C8	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	8	D8	
SI	DE	
SO	A2	
CMDR	1D	8	E7	
SI	0	0	
SO	A2	18	
CMDR	C0	
SI	0	
SO	22	
CMDW	95	
SI	4	7	0	1	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	8E	
SI	A	4	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	8	E7	
SI	18	
SO	A2	
success init!
[LoRaWAN] Resuming previous session ... CMDR	1D	8	E7	
SI	0	0	
SO	A2	18	
CMDR	C0	
SI	0	
SO	22	
CMDW	95	
SI	4	7	0	1	
SO	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	8E	
SI	10	4	
SO	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	8	E7	
SI	18	
SO	A2	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	40	
SI	34	44	
SO	A2	A2	
CMDR	11	
SI	0	0	
SO	A2	1	
CMDR	C0	
SI	0	
SO	22	
CMDR	1D	7	36	
SI	0	0	
SO	A2	D	
CMDR	C0	
SI	0	
SO	22	
CMDW	D	7	36	
SI	D	
SO	A2	
CMDW	8C	
SI	0	8	0	FF	1	0	
SO	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
failed, code -1101
````

nmaas87, 2023-11-04:
Thanks a lot, another update.
I am currently using PR https://github.com/jgromes/RadioLib/pull/867 which does include my PR #865 / PR #866 and hopefully soonish also the PR #868 - with all these changes and the general improvements @StevenCellist has done, I think everything should be working now using his PR (as soon as he includes PR #868 and removes one stray function :)).

The only thing that is still open for me is the part of the power-ing of the antenna switch - any ideas on how to do this better?:

DIO2 controls the "direction" (e.g. TX or RX), GPIO25 just powers the switch.
However, the switch should only get enabled if we actually want to RX or TX to keep the power profile as low as possible. Do you have any suggestions for that instead of a gloabl:

pinMode(25, OUTPUT);
digitalWrite(25, HIGH); 

Thanks a lot,
Nico

jgromes, 2023-11-04:
Pretty behind the curve on this one, but regarding the RF switching, you can use the `setRfSwitchPins` method. By default it assumes there are separate RX and TX enable pins, so you will have to use a custom RF switch matrix, something like this:

```c++
static const uint32_t rfswitch_pins[] = { 25, RADIOLIB_NC, RADIOLIB_NC };
static const Module::RfSwitchMode_t rfswitch_table[] = {
  {MODE_IDLE,  { LOW }},
  {MODE_RX,    { HIGH }},
  {MODE_TX,    { HIGH }},
  END_OF_MODE_TABLE,
};

(...)

radio.setRfSwitchTable(rfswitch_pins, rfswitch_table); // must be called before begin!
radio.begin();
```

nmaas87, 2023-11-04:
Thanks so much @jgromes - this worked a treat. I was actually thinking about using setRfSwitchPins - but then I realized that they will be probably trying to switch the same pin on and off - so that might be an issue - and then I started to go through the documentation and just hit the rfswitch_table as well 💯 .

Short remark, the setRfSwitchTable documentation ( https://jgromes.github.io/RadioLib/class_module.html#ac308fa817a5c36c5dc724a0d15cefd4d ) still includes the ````Module::END_OF_MODE_TABLE,```` ending, which throws an error.

Thanks a lot I have all my issues now resolved, I am looking forward to the #867 and when this is there, the RAK11300 will be possible with the RadioLib, awesome! 👍🏻 

As soon as this PR is through and I got the RAK11300 included in the Arduino-Pico framework I could create some LoRaWAN examples specific to the module if wanted :)

closed 2023-11-12 by jgromes