jgromes/RadioLib #869: Problems with LoRaWAN using SX1262 on Helium Network

MrTalon63, 2023-11-05:
**Describe the bug**
After successful OTAA, restarting results in the microcontroller producing -12 (Invalid frequency) error codes. After enabling debug mode I can see that it tries to set the frequency to 0MHz `[LoRaWAN] Sending uplink packet ... Channel frequency UL = 0.000000 MHz`

**To Reproduce**
```cpp
#include <RadioLib.h>
SX1262 radio = new Module(2, 5, 22, 4);
LoRaWANNode node(&radio, &EU868);

void setup() {
  Serial.begin(115200);

  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  uint64_t joinEUI = 0x111111111111111;
  uint64_t devEUI = 0x111111111111111;

  uint8_t nwkKey[] = { 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11};
  uint8_t appKey[] = { 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11};

  Serial.print(F("[LoRaWAN] Attempting over-the-air activation ... "));
  state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  Serial.print(F("[LoRaWAN] Sending uplink packet ... "));
  int state = node.uplink("test", 10);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  while (true);
}
```

**Expected behavior**
After reboot, the microcontroller should retrieve all of the data from EEPROM and use all of the usable channels, in this case 8 channels, just like on TTN.

**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type: SX1262
 - Arduino IDE version: 2.2.1
 - Library version: 6.2.0
 - EEPROM HEX dump:
```
22,0,0,0,39,0,0,48,1,0,0,0,1B,10,AD,12,3,15,A0,21,82,82,10,A5,8B,CB,29,33,25,12,3C,90,D4,1F,6E,1,69,A4,74,8E,4F,8D,30,D,71,C7,65,4C,D4,1F,6E,1,69,A4,74,8E,4F,8D,30,D,71,C7,65,4C,D4,1F,6E,1,69,A4,74,8E,4F,8D,30,D,71,C7,65,4C,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,FF,FF,FF,FF
```

jgromes, 2023-11-05:
There's a currently open pull request (#867) which includes significant overhauls of my original (quite ... suboptimal) LoRaWAN implementation. Could you try if the problem persists with that PR?

MrTalon63, 2023-11-05:
After replacing the library with one from PR and tweaking the code to make it work, it seems to set frequencies properly but completely misses TX windows, resulting in `-1108` error code.

MrTalon63, 2023-11-05:
Okay, it's working now. I have changed my code to send uplink every 2 minutes, and it seems like it misses the (?) first TX window just after reboot, but after the 2-minute wait, it sends the packet perfectly. I'm also not very knowledgeable when it comes to LoRaWAN, which may be perfectly normal.

StevenCellist, 2023-11-06:
There's something about the first uplink indeed; one of the 'timers' appears to be still running after the join/restore. Adding a delay(5000) at the end of the setup() is the easiest fix until I find out what actually makes this happen. Glad to hear it's working fine!

MrTalon63, 2023-11-06:
So if I restore the session at first and then wait for an example for a GPS fix, it should work right away?

StevenCellist, 2023-11-06:
I hope to address the bug such that you don't need to wait those few seconds, but for now, yes! Restore first in the setup(), then wait for a GPS fix - that will most likely take longer than the timer to run out.

MrTalon63, 2023-11-06:
Also last thing about LoRaWAN, how is the Spreading Factor chosen, and can it be changed manually by changing it on the physical interface?

StevenCellist, 2023-11-06:
In the PR code, you can set the SF during join (as a fifth argument) or after that by selecting .setDatarate(). 
Note that these arguments expect a Datarate (e.g. between 0 and 5 for EU868 band), not an SF.
Speaking about this, I should actually implement a check that verifies that a valid value is chosen.
Currently also investigating the cause of the Tx window problem you encountered.

MrTalon63, 2023-11-09:
Here is a small update regarding PR #867. The latest commits seem to fix another problem when using the library on the Helium network when packets would be labelled as late, from what I know it might have something to do with the fcnt counter.

StevenCellist, 2023-11-09:
Hey @MrTalon63, thank you for checking out and that's great to hear! I also encountered that notification on one of my earlier commits, but haven't been able to reproduce either in the latest version(s). I do not know what that "late" signified as I could not find any sources on Mr. Google, but happy to hear that your bug is resolved.

MrTalon63, 2023-11-09:
I just noticed that .setDatarate() method is missing. Is it on purpose?

StevenCellist, 2023-11-09:
Please explain: where do you think it is missing?

MrTalon63, 2023-11-09:
When using it with the previous code I got a compilation error
```cpp
error: 'class LoRaWANNode' has no member named 'setDatarate'; did you mean 'dataRate'?
   node.setDatarate(5);
        ^~~~~~~~~~~
        dataRate

exit status 1

Compilation error: 'class LoRaWANNode' has no member named 'setDatarate'; did you mean 'dataRate'?
```

StevenCellist, 2023-11-09:
The function `setDatarate()` is new in the latest commit, so you would need to copy the complete code from the latest commit.

jgromes, 2023-11-12:
@MrTalon63 #867 was just merged - my understanding is that it also resolves this issue. Could you please confirm?

MrTalon63, 2023-11-12:
I think yes, all of the issues have been resolved by the latest commit. If any other arises, I'll just open another issue.

closed 2023-11-12 by MrTalon63