jgromes/RadioLib #870: CC1101 begin() fails on ESP32 with code -16

xNombre, 2023-11-05:
Hello,
I am using ESP32 connected with CC1101 via VSPI. I cannot pass begin() without -16 error. It looks like SPI write is not successful.
What I tested already: tried connection to rfm95 with same setup - succeeded, tried connecting my cc1101 to arduino uno and executing begin() - ok.

Current code:
`#include <RadioLib.h>

SPIClass spi(VSPI);
CC1101 wmbus = new Module(22, 26, RADIOLIB_NC, RADIOLIB_NC, spi, SPISettings(2000000, MSBFIRST, SPI_MODE0));

void setup()
{
	Serial.begin(115200);
	spi.begin();

	int state = wmbus.begin(868.00);
}`

Console output:
`RadioLib Debug Info
Version:  6.2.0.0
Platform: ESP32
Compiled: Nov  5 2023 21:41:41

R       71      14
M       CC1101
CMD     W       30      00
CMD     W       36      0E
R       75      1
R       18      4
W       18      14
R       18      4
R       18      4
R       18      4
R       18      4
R       18      4
R       18      4
R       18      4
R       18      4

address:        0x18
bits:           5 4
value:          0x10
current:        0x4
mask:           0x30
new:            0x14
read:           0x4
R       18      4
W       18      4
R       18      4
failed, code -16`

Do you have any ideas for that?

jgromes, 2023-11-06:
Seems like it has issues enabling automated calibration, which is quite strange indeed. Could you try lowering the SPI clock speed, e.g. to 1 MHz?

Also:

> tried connecting my cc1101 to arduino uno and executing begin() - ok.

Arduino Uno is a 5V logic board, the CC1101 most likely has 3.3V logic voltage. Did you use a logic level shifter?

xNombre, 2023-11-10:
Hi, after lots of testing it worked eventually.
It looks like goldpin connections are the problem, cables need to be very tight and breadboard is a big no no - lesson for everyone.

Answering your question about Arduino: yes, I connected it directly, yes I know it was stupid but I was also tired of hours of testing. Also, there are some articles available where lls is not used, no comment. Surprisingly, cc1101 is still alive.

I can also confirm https://github.com/jgromes/RadioLib/issues/839 is alive and rocking.

closed 2023-11-10 by xNombre