jgromes/RadioLib #873: RP2040 w/ RFM95 Quit Initializing

loganfarrow, 2023-11-07:
**Sketch that is causing the module fail**

```c++
#include <RadioLib.h>

// Define the pin mapping for the RFM95 LoRa module
#define RFM95_CS    16  // Chip select pin
#define RFM95_RST   17  // Reset pin
#define RFM95_IRQ   21  // Interrupt pin, connected to DIO0
#define RFM95_GPIO  22  // Additional GPIO, connected to DIO1 

SX1276 radio = new Module(RFM95_CS, RFM95_IRQ, RFM95_RST, RFM95_GPIO);

void setup() {

  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(RFM95_RST, OUTPUT);
  digitalWrite(RFM95_RST, HIGH);

  Serial.begin(115200);
  while (!Serial) delay(1);
  delay(100);

  Serial.println("Feather LoRa RX Test!");

  // manual reset
  digitalWrite(RFM95_RST, LOW);
  delay(10);
  digitalWrite(RFM95_RST, HIGH);
  delay(10);

  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    // no error
    Serial.println(F("Initialization successful!"));
  } else {
    // some error occurred
    Serial.print(F("Initialization failed, code "));
    Serial.println(state);
    while (true);
  }

}

void loop() {
  Serial.print("Sending message...");
  // Send a string "Hello, World!".
  int state = radio.transmit("Hello, World!");
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println("success!");
  } else {
    Serial.print("failed, error code: ");
    Serial.println(state);
  }

  // Wait for a bit before sending the next message.
  delay(1000);
}

```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
Feather LoRa RX Test!

RadioLib Debug Info
Version:  6.2.0.0
Platform: Raspberry Pi Pico (unofficial)
Compiled: Nov  7 2023 16:32:38

R	42	0	
SX127x not found! (1 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0000, expected 0x0012
R	42	FF	
SX127x not found! (2 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (3 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (4 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (5 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (6 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (7 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (8 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (9 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
R	42	FF	
SX127x not found! (10 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x00FF, expected 0x0012
No SX127x found!
Initialization failed, code -2

**Additional info (please complete):**
 - MCU: RP2040
 - Link to Arduino core: 
 - Wireless module type - Adafruit says SX1272 but worked as SX1276
 - Arduino IDE version  2.2.1
 - Library version 6.2.00

For some reason, this exact code was working, and then after I connected some other components over i2c (IMU, BME680) I am now getting this error when switching back to my original script. Any chance I could get some help to fix it?

jgromes, 2023-11-08:
From the debug output it looks like there is nothing on the SPI bus, and since the transceiver and the RP2040 are on the same board, I don't think it's a wiring issue. Is the power supply onboard the Feather board powerful enough to power all the extra components?

loganfarrow, 2023-11-08:
The issue persists even after disconnecting the extra components and trying again. The only way I got the sketch to work again was reinstalling all of my libraries and the Arduino IDE + factory reset on the board. Any ideas on what might have caused the issue?

jgromes, 2023-11-11:
That's odd - then I would guess there is some common peripheral that handles both I2C and SPI communication on the RP2040 and for some reason it's not being initialized correctly. Unfortunately I don't think there's much I can do, the library handled the lack of replies from the radio correctly. It seems like an issue with the platform itself.

Feel free to reopen if more information becomes available.

closed 2023-11-11 by jgromes