jgromes/RadioLib #874: Fails to Compile due to Breaking Changes in Arduino ESP32 Library 3.0.0 Alpha

coolacid, 2023-11-10:
**IMPORTANT: Check the wiki**
Checked - Not found. No issue found either. 

**Describe the bug**
Arduino-ESP32 Library is making API breaking changes in the new 3.0.0 version.

```
Arduino: 1.8.14 Hourly Build 2021/04/08 03:25 (Linux), Board: "ESP32S2 Dev Module, FTDI Adapter, Disabled, Disabled, Disabled, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

/home/coolacid/Arduino/libraries/RadioLib/src/ArduinoHal.cpp: In member function 'virtual void ArduinoHal::tone(uint32_t, unsigned int, long unsigned int)':
/home/coolacid/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:144:7: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  144 |       ledcAttachPin(pin, RADIOLIB_TONE_ESP32_CHANNEL);
      |       ^~~~~~~~~~~~~
      |       ledcAttach
/home/coolacid/Arduino/libraries/RadioLib/src/ArduinoHal.cpp: In member function 'virtual void ArduinoHal::noTone(uint32_t)':
/home/coolacid/Arduino/libraries/RadioLib/src/ArduinoHal.cpp:177:5: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  177 |     ledcDetachPin(pin);
      |     ^~~~~~~~~~~~~
      |     ledcDetach
exit status 1
Error compiling for board ESP32S2 Dev Module.
``` 

**To Reproduce**
Download and install ESP32 3.0.0 alpha code into Arduino IDE. Try and compile APRS Mic-E Example. 

**Expected behavior**
Compile Failure

**Screenshots**
N/A

**Additional info (please complete):**
Breaking Changes Documentaion: https://github.com/espressif/arduino-esp32/blob/master/docs/source/migration_guides/2.x_to_3.0.rst#ledc

jgromes, 2023-11-11:
Which version of the library are you using? I think this was already addressed in 039fa0fc59af9e367ea8c4b8429466f297f7d7ce, however, there was no release published since. At the moment a big PR is being prepared and once that's done, a new release will be published.

MonkeybarWooWoo, 2023-11-22:
I get this compilation failure too when using   jgromes/RadioLib @ ^6.2.0

jgromes, 2023-11-22:
@MonkeybarWooWoo see my previous comment - new release is still being prepared, that's why this issue is still open.

MonkeybarWooWoo, 2023-11-22:
Thanks very much Jan.

jgromes, 2023-11-29:
Closing since 6.3.0 released with this fix.

closed 2023-11-29 by jgromes