jgromes/RadioLib #876: ESP32S3-wroom-2 and RFM22b-s2 SPI error -16

tresler, 2023-11-16:
Hi again. I am trying connecting ESP32S3-wroom-2 and RFM22b-s2. I don't know if is it issue or I am doing something bad, but I have _code -16_ (RADIOLIB_ERR_SPI_WRITE_FAILED) and I don't know why. I was trying release and also actual dev version of RadioLib, also as arduino-ESP32 v2.0.14 and v3 alpha2 and also today actual dev and without luck. Do you have any idea or information for me please? 

**My code:**
`// include the library
#include <RadioLib.h>

// RFM22:
#define nSEL 40
#define nIRQ 38
#define SDN 39
#define SDO 21
#define SDI 1
#define SCK 3
Si4432* radio;
SPIClass* rspi = NULL;
Module* modul;

void setup() {
  Serial.begin(115200);
  delay(3000);

  rspi = new SPIClass(HSPI);
  rspi->begin(SCK, SDO, SDI);
  modul = new Module(nSEL, nIRQ, SDN, RADIOLIB_NC, *rspi);
  //modul = new Module(nSEL, nIRQ, SDN);
  radio = new Si4432(modul);
  // initialize RFM22 with default settings
  Serial.print(F("[RFM22] Initializing ... "));
  int state = radio->begin(868.0);

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }`

jgromes, 2023-11-17:
SPI write failure usually indiciates some hardware issue. Can you post the debug mode output, as well as information about your hardware setup (as asked for by the issue template)?

tresler, 2023-11-18:
Thank you for replay, sorry for late response. Here is it:

`13:45:01.230 -> [   331][V][esp32-hal-periman.c:150] perimanSetBusDeinit(): Dein=========== Before Setup Start ===========
13:45:01.230 -> Chip Info:
13:45:01.230 -> ------------------------------------------
13:45:01.230 ->   Model             : ESP32-S3
13:45:01.230 ->   Package           : 0
13:45:01.230 ->   Revision          : 1
13:45:01.230 ->   Cores             : 2
13:45:01.230 ->   Frequency         : 240 MHz
13:45:01.230 ->   Embedded Flash    : No
13:45:01.230 ->   Embedded PSRAM    : No`

jgromes, 2023-11-18:
I meant the RadioLib debug: https://github.com/jgromes/RadioLib/wiki/Debug-mode

Sorry for not being totally clear (though that same link is in the issue template ...)

tresler, 2023-11-18:
Sorry for that. This is right? 

`RadioLib Debug Info
Version:  6.2.0.0
Platform: ESP32
Compiled: Nov 18 2023 15:20:29

R	1	6	
M	Si443x
W	7	80	
R	3	0	3	
R	7	0	
W	7	1	
R	7	0	
R	7	0	
R	7	0	
R	7	20	
R	7	3	
R	7	1	
W	6	0	
R	69	14	
W	69	74	
R	69	A	
R	69	64	
R	69	47	
R	69	2
R	69	0	
R	69	0	
R	69	0
R	69	18	
R	69	26	
R	69	8D	
R	69	C	
R	69	8	
R	69	2D	
R	69	0
R	69	0	
R	69	0	
R	69	0	
R	69	0	
R	69	0	
R	69	FF	
R	69	FF	
R	69	0	

address:	0x69
bits:		6 5
value:		0x60
current:	0x14
mask:		0x60
new:		0x74
read:		0x0
failed, code -16`

jgromes, 2023-11-18:
That's the correct debug. Looks like your Si4432 is behaving qutie erractically, returning what seems to be random data. I would suggest checking the hardware, pintou, wiring, power etc.

tresler, 2023-11-18:
Yes, I am stupid. It didn't look like it, but one pin was soldered wrong. Now it is going well. Thank you very much for your help and please accept my apologies for the wasted time.

jgromes, 2023-11-18:
No worries, I'm glad you were able to resolve the problem!

closed 2023-11-18 by jgromes

tresler, 2023-11-20:
Unfortunately now I have problem with transmit. I have good SPI connect code after radio->begin(868.0);
`[RFM22] Initializing ... 
RadioLib Debug Info
Version:  6.2.0.0
Platform: ESP32
Compiled: Nov 20 2023 14:34:21

R	1	6	
M	Si443x
W	7	80	
R	3	0	0	
R	7	1	
W	7	1	
R	7	1	
W	6	0	
R	69	20	
W	69	60	
R	69	60	
R	33	22	
W	33	2	
R	33	2	
R	B	0	
W	B	12	
R	B	12	
R	C	0	
W	C	15	
R	C	15	
R	32	C	
W	32	0	
R	32	0	
R	70	C	
W	70	2C	
R	70	2C	
W	6E	27	
W	6F	52	
R	1C	1	
R	1C	1	
R	70	2C	
0
0
0
833.33	833	341
10066	2752
65535	FFFF
R	21	1	
W	21	61	
R	21	61	
R	20	64	
W	20	41	
R	20	41	
R	21	61	
W	21	60	
R	21	60	
R	22	47	
W	22	27	
R	22	27	
R	23	AE	
W	23	52	
R	23	52	
R	24	2	
W	24	7	
R	24	7	
R	25	8F	
W	25	FF	
R	25	FF	
R	71	0	
W	71	0	
R	71	0	
W	72	8	
R	1C	1	
W	1C	96	
R	1C	96	
R	1C	96	
R	1C	96	
R	70	2C	
1
1
0
1250.00	1250	4E2
6710	1A36
33	21
R	21	60	
W	21	80	
R	21	80	
R	20	41	
W	20	E2	
R	20	E2	
R	21	80	
W	21	80	
R	21	80	
R	22	27	
W	22	1A	
R	22	1A	
R	23	52	
W	23	36	
R	23	36	
R	24	7	
W	24	0	
R	24	0	
R	25	FF	
W	25	21	
R	25	21	
R	34	8	
W	34	4	
R	34	4	
R	35	2A	
W	35	12	
R	35	12	
R	7	1	
W	7	1	
R	7	1	
R	33	2	
W	33	2	
R	33	2	
W	36	12	AD	
R	71	0	
W	71	2	
R	71	2	
R	71	2	
W	71	22	
R	71	22	
R	7	1	
W	7	1	
R	7	1	
R	70	2C	
W	70	28	
R	70	28	
R	7	1	
W	7	1	
R	7	1	
R	70	28	
W	70	28	
R	70	28	
R	33	2	
W	33	2	
R	33	2	
R	3E	0	
W	3E	40	
R	3E	40	
M	Si4432
R	7	1	
W	7	1	
R	7	1	
R	75	75	
W	75	73	
R	75	73	
R	76	BB	
W	76	64	
R	76	64	
R	77	80	
W	77	0	
R	77	0	
R	2A	0	
W	2A	28	
R	2A	28	
R	6D	18	
W	6D	1B	
R	6D	1B	
success!
R	7	1	
W	7	1	
R	7	1	
R	33	2	
W	33	6	
R	33	6	
W	36	92	10	19	1	
Run with code: 1`

But unfortunately if I want send String or Array I have code:
`R	7	FF	
W	7	1	
R	7	FF
same XX lines

address:	0x7
bits:		7 0
value:		0x1
current:	0xFF
mask:		0xFF
new:		0x1
read:		0xFF
failed, code -16`

I have check twice my pinout and also have connected antenna. My code is easy like in example:
`bool radio_send(byte device, int number_of_value, byte values[]) {
  bool succest;
  byte send_array[number_of_value+1];
  Serial.print("num data: ");
  Serial.println(number_of_value);

  send_array[0] = device;
  for (int i = 0; i < number_of_value; i++) {
    send_array[i+1] = values[i];
  }
  /*
  String send_string = String(device);

  for (int i = 0; i < number_of_value; i++) {
    send_string += "," + String(values[i]);
  }
  Serial.println(send_string);

  int state = radio->transmit(send_string);
  */
  int state = radio->transmit(send_array, number_of_value+1);

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));
    succest = true;

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F(" too long!"));
    succest = false;

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F(" timeout!"));
    succest = false;

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
    succest = false;
  }

  return succest;
}`

jgromes, 2023-11-21:
Can you also post the complete debug output? It seems like the device stops responding at some point, only returning 0xFF .

tresler, 2023-11-21:
Of course, here:
[radiolib_debug_code.txt](https://github.com/jgromes/RadioLib/files/13431493/radiolib_debug_code.txt)

jgromes, 2023-11-22:
Looking at the debug output, whatever caused this happened before entering the `radio_send` function. You haven't posted that code so I can't comment much, but it seems that the SPI to the radio doesn't work at all. So I agree that if there is a second SPI instance it could somehow interfere. Though that is very ESP32-specific.