jgromes/RadioLib #877: Reasons behind "RADIOLIB_ERR_TX_TIMEOUT"

Bushpulbek, 2023-11-21:
Hello, everybody.

Hope you're doing well

I've been trying to play around with SX1276 recently, and LoRa transmission/reception was OK.
However, when switching to FSK I was trying to use the examples from "**SX127x_FSK_Modem.ino**"

Carrier frequency is **960 MHz**

The code is given below:
```
void setup(){
  // initialize SX1276 FSK modem with default settings
  Serial.print(F("[SX1276] Initializing ... "));
  int state = radio.beginFSK(960.0);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }
}

void loop(){
  // NOTE: FSK modem maximum packet length is 63 bytes!
  int state = radio.transmit("Hello World!");

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("[SX1276] Packet transmitted successfully!"));
  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    Serial.println(F("[SX1276] Packet too long!"));
  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    Serial.println(F("[SX1276] Timed out while transmitting!"));
  } else {
    Serial.println(F("[SX1276] Failed to transmit packet, code "));
    Serial.println(state);
  }
}
```

I am constantly receiving the message "[SX1276] Timed out while transmitting!" in Terminal => RADIOLIB_ERR_TX_TIMEOUT error code.

I do not comprehend the meaning of "Timed out waiting for transmission finish". What factors may stand behind it?

jgromes, 2023-11-21:
First of all, please include the information requested in the issue template. What is the hardware you are using? What is the library version? What is the Arduino core version? Which Arduino are you using? Etc.

The SX127x signals end of transmission on the DIO0 pin, so the usual reason for this error is bad connection on DIO0.

Bushpulbek, 2023-11-22:
First of all, my apologies for not following the template. Edited comment.
Can't share the schematic, though(

Thank you very much for your comment about DIO0. That really made everything clear

jgromes, 2023-11-22:
I would start by examining the DIO0 connection, there should a low-to-high transition at every packet end. I would also examine by receiver or an SDR if something is being transmitted. The error only means that while waiting for the end of transmission, the program timed out waiting for the DIO0 signal. That doesn't mean the packet was not sent though.

Bushpulbek, 2023-11-23:
Thank you. Connected DIO0 to MCU's GPIO pin => everything's OK now. Got **RADIOLIB_ERR_NONE** status

closed 2023-11-23 by Bushpulbek