jgromes/RadioLib #879: External EEPROM support

MrTalon63, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
As not every MCU has EEPROM memory, some have to rely on using flash storage as one or can't use it with this library right now. When it comes to emulating EEPROM memory on flash, it comes at a cost of much smaller durability (around 100000 cycles when I searched for flash memory durability that is used in ESP32 MCUs), compared to purpose-built ICs like the AT24C08, which boasts even 1 million write cycles.

**Describe the solution you'd like**
While a plug-and-play solution would be great, there are too many EEPROM chips out there, some of which have different requirements regarding block sizes, etc. I think that an easier method would be to expose the functionality by letting each end-user access `readPersistentStorage` and `writePersistentStorage` so they could define their own functions. I'm not entirely sure about the viability of such a method and it might be outrageously unsafe or difficult to implement properly.

**Describe alternatives you've considered**
Currently, the only alternative I see is to modify `ArduinoHal.cpp` functions to personal needs.

jgromes, 2023-11-22:
Couple of points:

1. EEPROM is only required for saving LoRaWAN sessions. If you don't use LoRaWAN or don't need to save session state between power cycles, you can use the library without EEPROM just fine. Though I have to note that the LoRaWAN specification requires that some variables must be persistent.
2. It is true that the cycle count of flash is typically much smaller than EEPROM - it is a different technology after all. However, RadioLib LoRaWAN implementation is aware of that, and implements wear leveling that limits the number of persistent memory erase cycles needed, thus improving the lifetime. In real world use, I have yet to see a device that would fail due to excessive flash cycling (outside of purpose testing).
3. Both `read` and `writePersistentStorage` are part of the HAL - both are virtual methods that can be overriden. By default, they are overriden by the implementation from `ArduinoHal` class, but nothing is stopping you from writing your own HAL. See the non-Arduino examples.

jgromes, 2023-12-07:
Closed due to inactivity

closed 2023-12-07 by jgromes